platformio/platformio-vscode-ide #884: Installation%20Manager

erkuo521, 2019-07-10:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.36.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17763%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20Error%3A%20If%20you%20use%20Antivirus%2C%20it%20can%20block%20PlatformIO%20Installer.%20Try%20to%20disable%20it%20for%20a%20while.%0A%0AVirtualenv%20Create%3A%20Traceback%20(most%20recent%20call%20last)%3A%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5C.cache%5Ctmp-118288tFryEw6OnEW%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%202462%2C%20in%20%3Cmodule%3E%0D%0A%20%20%20%20main()%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5C.cache%5Ctmp-118288tFryEw6OnEW%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%20762%2C%20in%20main%0D%0A%20%20%20%20symlink%3Doptions.symlink%2C%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5C.cache%5Ctmp-118288tFryEw6OnEW%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%201015%2C%20in%20create_environment%0D%0A%20%20%20%20install_wheel(to_install%2C%20py_executable%2C%20search_dirs%2C%20download%3Ddownload)%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5C.cache%5Ctmp-118288tFryEw6OnEW%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%20968%2C%20in%20install_wheel%0D%0A%20%20%20%20call_subprocess(cmd%2C%20show_stdout%3DFalse%2C%20extra_env%3Denv%2C%20stdin%3DSCRIPT)%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5C.cache%5Ctmp-118288tFryEw6OnEW%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%20812%2C%20in%20call_subprocess%0D%0A%20%20%20%20env%3Denv%2C%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5Cpython27%5CLib%5Csubprocess.py%22%2C%20line%20390%2C%20in%20__init__%0D%0A%20%20%20%20errread%2C%20errwrite)%0D%0A%20%20File%20%22C%3A%5CUsers%5Cchenerkuo%5C.platformio%5Cpython27%5CLib%5Csubprocess.py%22%2C%20line%20640%2C%20in%20_execute_child%0D%0A%20%20%20%20startupinfo)%0D%0AWindowsError%3A%20%5BError%205%5D%20%0D%0A%0APlease%20use%20the%20*system*%20python%20to%20run%20this%20script%0D%0AInstalling%20setuptools%2C%20pip%2C%20wheel...%0D%0A%0D%0A%20%20Error%20%5BError%205%5D%20%20while%20executing%20command%20C%3A%5CUsers%5Cchenerkuo%5C.platformio%5Cpenv%5CScripts%20-%20setuptools%20pip%20wheel%0D%0A...Installing%20setuptools%2C%20pip%2C%20wheel...done.%0D%0A%0A%09at%20t.value%20(c%3A%5CUsers%5Cchenerkuo%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A25605)%0A%60%60%60%0A

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

HazemRagabElsayed, 2024-07-29:
[image]

Another problem that need to be solved

ExploiTR, 2025-06-01:
> [image]
> 
> Another problem that need to be solved

Just disable AV and do it.