platformio/platformio-vscode-ide #886: Start%20PIO%20Home%20Server

pickmylight, 2019-07-10:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.36.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Error%3A%20Detected%20unknown%20package%20'tool-pioplus'%0D%0A%0A%09at%20P.Promise%20(c%3A%5CUsers%5Cschmipa%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A15270)%0A%09at%20ChildProcess.h%20(c%3A%5CUsers%5Cschmipa%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(c%3A%5CUsers%5Cschmipa%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A962%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A251%3A5)%0A%60%60%60%0A

pbolduc, 2019-07-10:
@pickmylight do you really want help or are you just spamming the issues?  You provide no information other.  Do not post url encode values.  If I was the moderator, I could just close your issue.  What **your** garbage above says

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.36.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Error: Detected unknown package 'tool-pioplus'

	at P.Promise (c:\Users\schmipa\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:15270)
	at ChildProcess.h (c:\Users\schmipa\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\schmipa\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-07-13:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/837

closed 2019-07-13 by ivankravets