platformio/platformio-vscode-ide #888: PIO Settings in VSCode

sstaub, 2019-07-10:
There are some issues in the settings:
- I have to restart or reload to make changes executed, it should work immediately if possible,
like other vscode setting also do.
- Sometimes I have to force the setting by clicking off and on again, this happens with e.g.
'Use Development PIOCore'
- There should an option for setting automatic start for PIO Home, this setting was available in earlier versions. In the moment this only possible in PIO Home Tab, but when there is no connection or other problem this is annoying because VSCode try to start the PIO Home tab every time I start VSCode

closed 2019-07-15 by ivankravets

ivankravets, 2019-07-15:
Please re-test with https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.0-beta.1

sstaub, 2019-07-16:
The additional setting works. :)

ivankravets, 2019-07-16:
I think maybe is correct to don’t pre start PIO Home by default. Developers will see some delay on the first start of PIO Home. But the next starts should work fast. 

What do you think?

sstaub, 2019-07-16:
I think PIO Home shouldn't start by default.

ivankravets, 2019-07-16:
Ok, we disabled it. The option was changed to `platformio-ide.disablePIOHomeStartup`

commit 2adb2e44 (2019-07-15): Added new setting `platformio-ide.disableAutostartPIOHomeServer` which disables automatic starting of PIO Home server // Resolve #888
commit 9ebe5b80 (2019-07-16): Option to disable PIO Home at startup // Resolve #888
commit 7199b884 (2019-07-16): Option to disable PIO Home at startup // Resolve #888