platformio/platformio-vscode-ide #889: Installation%20Manager

searayca, 2019-07-10:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.36.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17763%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20PIP%20Core%3A%20If%20you%20have%20antivirus%2Ffirewall%2Fdefender%20software%20in%20a%20system%2C%20try%20to%20disable%20it%20for%20a%20while.%20%0A%20C%3A%5Cmsys%5C1.0%5Chome%5C.platformio%5Cpython27%5Cpython.exe%3A%20No%20module%20named%20pip%0D%0A%0A%09at%20f.value.i.runCommand%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A26495)%0A%09at%20ChildProcess.h%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A962%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A251%3A5)%0A%60%60%60%0A

pbolduc, 2019-07-10:
@searayca do you really want help or are you just spamming the issues?  You provide no information other.  Do not post url encode values.  If I was the moderator, I could just close your issue.  What **your** garbage above says

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.36.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 C:\msys\1.0\home\.platformio\python27\python.exe: No module named pip

	at f.value.i.runCommand (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:26495)
	at ChildProcess.h (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

searayca, 2019-07-10:
We would really love help.

We are on windows 10 professional. We install platform io. It spins for 2
minutes, gives error message and repeats.

We have disabled Comodo and bit defender.

Please advise.

pbolduc, 2019-07-11:
How did you submit your request because the initial request is very difficult to understand.  It is all URL encoded.

searayca, 2019-07-11:
I am trying to install platform.io on my Windows 10 workstation.

It starts to install, wants to restart the ide, and then installs
platform.io again. Over and over and over.

I have tried disabling firewalls and antivirus.

Installed clean on my Mac first time.

Any suggestions?

Greg

demosthien, 2019-07-13:
> How did you submit your request because the initial request is very difficult to understand. It is all URL encoded. Your decoded error message says pip is not installed. Can you confirm pip is installed?

That's how PIO self-submits its issues... In URL code, I have mentioned this for the Devs in issue #809 but they've not responded yet.

pbolduc, 2019-07-13:
> That's how PIO self-submits its issues... In URL code, I have mentioned this for the Devs in issue #809 but they've not responded yet.

Oh, I see.  I am a just an end user myself.  I have noticed other issues with the text boxes in the library manager stop working after navigating away and back.

searayca, 2019-07-13:
I would sure like to use platform io, but can’t install. It tries over and
over and over and over and over. I have turned off ALL firewalls.

Greg

pfeerick, 2019-07-13:
> That's how PIO self-submits its issues... 
@demosthien 

I just went down the rabbit hole and pinned down the code I think to be responsible for that URI encoding gobbledegook in platformio/platformio-node-helpers/issues/7, which will make it more liekly to be resolved or at least commented on. ;)

demosthien, 2019-07-13:
Thanks @pfeerick !! Hopefully doing that will prompt a fix.

The URL code auto-submit issue is, I think, a big reason why so many duplicates are being submitted. People click the submit button and then it looks like garbage so they don’t try to find if there is a fix & just leave it.

ivankravets, 2019-07-13:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/815

closed 2019-07-13 by ivankravets