platformio/platformio-vscode-ide #892: can't type project name on my mac

morestart, 2019-07-12:
can't type project name on my mac vscode, but i can use web interface type project name

pfeerick, 2019-07-13:
Sorry? Can you elaborate? Do you mean you can't enter a name for a new project in PIO Home when it's in a VSCode tab, but you can when you pop it out to a web browser?

morestart, 2019-07-13:
> Sorry... can you elaborate further? Do you mean you can't enter a name for a new project in PIO Home when it's in a VSCode tab, but you can when you pop it out to a web browser?

yeah, i don't know why, i can't copy demo code from pio home that use vscode, but i use chrome open the pio home, it works...

pfeerick, 2019-07-13:
When you say copy demo code - do you mean `open project` (1) or `project examples` (2) ? 

[image]

morestart, 2019-07-13:
no, it is libraries demo code

demosthien, 2019-07-18:
I can confirm this issue.

OSX: v10.14.3
VSCode:  v1.36.1
PIO: Home 2.2.0 Core 4.0.0

Occasionally when creating a "New Project" I have had the text entry boxes (this includes the search drop down menus as well) lock; no cursor appears and no text can be entered. The project can then not be created due to missing text such as a project name.

I have noticed typed text does sometime appear in a red box elsewhere on the screen (I have no idea what this other text box is for). See the attached image for a screenshot of this extra dislocated text next to the PIO Home button.

[image]

pfeerick, 2019-07-18:
I've been able to to get this to occur multiple times now, and seems more likely to occur if it's the first time PIO Home has opened in a VSCode session... closing the PIO Home tab and reopening it seems to 'cure' it. Nothing has shown up in the the vscode debug logs so far... and it's erratic ... Eureka! I think I just found the repeatable way to trigger it... 

1) Open PIO Home
2) click on some other element of VSCode - ie. sidebar icon, expand a collasped folder / project task
3) try to enter any text input into PIO Home - search project, search library, create new project... it won't have keyboard input focus :(

btw, the red box is to do with being able to find files in your workspace... if you have a file in one of the folders called 'debug' and you click on the file browser and just start typing 'debug' it will get highlighted for you you, and scrolled into view...

[image]

Erich11, 2019-07-23:
Can confirm this problem. I tried to convince my boss to use PlatformIO over Ardunio IDE. While I tried to show him the benefits of PIO this bug occured multiple times. Real show stopper.

[image]

pfeerick, 2019-07-23:
Another way to get it to work again is to click on the pio home tab, and bash the tab key on your keyboard several times until you see the buttons at the top of the pio page lighting up... then it seems it has keyboard focus again.

ivankravets, 2019-07-23:
This is a VSCode bug which was fixed in the 1.37.0 (has not been released yet). See related issues:
* https://github.com/microsoft/vscode/issues/76866
* https://github.com/James-Yu/LaTeX-Workshop/issues/1481

The workaround is to double-click on PIO Home tab.

Could someone re-test with VSCode Insider 1.37 https://code.visualstudio.com/insiders/?

Erich11, 2019-07-23:
Just tried it with the 1.37.0 insider build on W10 x64 as well as W7 x64. The issue persists.
Double-clicking on the PIO Home Tab works as mentioned.

pfeerick, 2019-07-23:
The workaround appears to be working on 1.36.1 (on Windows 10, at least)... double clicking the `PIO Home` tab is returning keyboard focus to PIO Home :)

ivankravets, 2019-07-24:
Let's wait for the fix on https://github.com/microsoft/vscode/issues/76863

Galaxy-Man, 2019-08-07:
Brilliant, thanks Ivan for your time in updating this, I dont know how you have time to do all these responses.

ivankravets, 2019-11-26:
The issue has been fixed in https://github.com/microsoft/vscode/issues/76863

Please update VSCode to 1.40.2 or above.

closed 2019-11-26 by ivankravets

pfeerick, 2019-11-26:
Funny you would mention that... I only just restarted VSCode to apply 1.40.2... didn't notice they'd bumped it to the October recovery build.  Nice to see the back of this one! :)