platformio/platformio-vscode-ide #896: Project env tasks no longer being generated

novirium, 2019-07-15:
Since the 4.0 update, I'm no longer getting any env-specific tasks being generated (the usual "build" and "upload" ones). I'm not familiar with the extension structure, but a clue might be the following error shown in the verbose VSCode output:

```
[1829:0715/093349.903144:INFO:CONSOLE(241)] "%c[Extension Host] %crunCommand", source: file:///usr/lib/code/out/vs/workbench/workbench.main.js (241)
[1829:0715/093350.102952:INFO:CONSOLE(241)] "%c[Extension Host] %cError: Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute 'get_home_dir'

	at r.value.Promise (/home/novirium/.vscode-oss/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:12126)
	at ChildProcess.h (/home/novirium/.vscode-oss/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5) %c(at e.value (/home/novirium/.vscode-oss/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:11653))", source: file:///usr/lib/code/out/vs/workbench/workbench.main.js (241)
```

novirium, 2019-07-15:
Digging around, platformio-node-helpers seems to still reference the Python `util.get_home_dir()` function from platformio-core [here](https://github.com/platformio/platformio-node-helpers/blob/7a2bf0a04fac1955dbb71bb9ddde0a84a74cd245/src/project/indexer.js#L115), which was removed from util.py in [this recent commit](https://github.com/platformio/platformio-core/commit/19725fec04302387af15c3942790bb26c98a7bfa).

ivankravets, 2019-07-15:
Please re-test with https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.0-beta.1

closed 2019-07-15 by ivankravets

novirium, 2019-07-17:
Can confirm, release 1.8 fixed this issue - thanks :+1: