platformio/platformio-vscode-ide #899: Installation%20Manager

searayca, 2019-07-16:
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 C:\msys\1.0\home\.platformio\python27\python.exe: No module named pip

	at f.value.i.runCommand (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:26495)
	at ChildProcess.h (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

All virus checks as also the firewalls. Another computer onsite has no problem installing.

PLEASE HELP!!!

Greg

%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.36.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17763%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20PIP%20Core%3A%20If%20you%20have%20antivirus%2Ffirewall%2Fdefender%20software%20in%20a%20system%2C%20try%20to%20disable%20it%20for%20a%20while.%20%0A%20C%3A%5Cmsys%5C1.0%5Chome%5C.platformio%5Cpython27%5Cpython.exe%3A%20No%20module%20named%20pip%0D%0A%0A%09at%20f.value.i.runCommand%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A26495)%0A%09at%20ChildProcess.h%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(C%3A%5CUsers%5Cgreg.ames%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A962%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A251%3A5)%0A%60%60%60%0A

ivankravets, 2019-07-16:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/900

closed 2019-07-16 by ivankravets

searayca, 2019-07-16:
Anybody seem to want to help fix this issue. Still never completes platform
io installation.

Help!!!

Greg