platformio/platformio-vscode-ide #900: Installation Manager

searayca, 2019-07-16:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 C:\msys\1.0\home\.platformio\python27\python.exe: No module named pip

	at f.value.o.runCommand (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:29748)
	at ChildProcess.h (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\greg.ames\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-07-16:
> C:\msys

It seems that PlatformIO uses this broken `msys` 's Python. It should not. Let me check a code.

ivankravets, 2019-07-16:
What does mean `C:\msys\1.0\home`? Could you uninstall all Python from the system and install manually a stable 3 version ? 
Also, remove after this step `C:\msys\1.0\home\.platformio` folder.

ivankravets, 2019-07-23:
Please reopen if you still need help

closed 2019-07-23 by ivankravets

searayca, 2019-07-26:
Sorry for the delay.

S:\>echo %path%
C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\MinGW\mingw32\bin;C:\Program Files
(x86)\Common Files\Oracle\Java\javapath;C:\Program Files
(x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program
Files\Intel\Intel(R) Management Engine
Components\iCLS\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program
Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program
Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files
(x86)\NVIDIA Corporation\PhysX\Common;C:\Program
Files\Intel\WiFi\bin\;C:\Program Files\Common
Files\Intel\WirelessCommon\;C:\Program
Files\Git\cmd;C:\Users\greg.ames\AppData\Local\Programs\Python\Python35\Scripts\;C:\Users\greg.ames\AppData\Local\Programs\Python\Python35\;C:\Users\greg.ames\AppData\Local\Microsoft\WindowsApps;C:\Users\greg.ames\AppData\Local\Programs\Microsoft
VS Code\bin

S:\>echo %PYTHONPATH%
%PYTHONPATH%

S:\>python --version
Python 3.5.0

S:\>pip --version
Traceback (most recent call last):
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 170, in _run_module_as_main
    "__main__", mod_spec)
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 85, in _run_code
    exec(code, run_globals)
  File
"C:\Users\greg.ames\AppData\Local\Programs\Python\Python35\Scripts\pip.exe\__main__.py",
line 5, in <module>
ImportError: cannot import name 'main'

S:\>pip
Traceback (most recent call last):
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 170, in _run_module_as_main
    "__main__", mod_spec)
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 85, in _run_code
    exec(code, run_globals)
  File
"C:\Users\greg.ames\AppData\Local\Programs\Python\Python35\Scripts\pip.exe\__main__.py",
line 5, in <module>
ImportError: cannot import name 'main'

S:\>pip list
Traceback (most recent call last):
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 170, in _run_module_as_main
    "__main__", mod_spec)
  File
"c:\users\greg.ames\appdata\local\programs\python\python35\lib\runpy.py",
line 85, in _run_code
    exec(code, run_globals)
  File
"C:\Users\greg.ames\AppData\Local\Programs\Python\Python35\Scripts\pip.exe\__main__.py",
line 5, in <module>
ImportError: cannot import name 'main'

S:\>

ivankravets, 2019-07-26:
Please uninstall Python 3.5 and install 3.7. Did it help?