platformio/platformio-vscode-ide #904: Start PIO Home Server

HenSpi, 2019-07-17:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at P.Promise (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:15396)
	at ChildProcess.h (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-07-23:
@HenSpi @lxf2020 , could you reproduce this issue again? Do you see any errors in VSCode > Help > Toggle Developer Tools > Console? If yes, please share here.

HenSpi, 2019-07-24:
Dear Ivan,
unfortunately yes, I can’t get the any of the Quick Access points from PIO menu. Every time it's opening a new tab and after a short time the same error appear. I copied it in here:
This error did not occur from the beginning. In the first weeks with PlatformIO, during the free trial period, the updates and access worked.

If there's anything else I can do, let me know. By the way, I think PlatformIO is the best IDE for Arduino, ESP, etc. there is so far. Many thanks for the development and your support!

Best Regards
  Hendrik Spilker

PS: I’m using MacOS Catalina in Version 10.15 Beta (19A501i)

ivankravets, 2019-07-25:
Please provide console errors from https://github.com/platformio/platformio-vscode-ide/issues/904#issuecomment-514157928

HenSpi, 2019-07-25:
Unfortunately I don't know what you mean. Please explain.

ivankravets, 2019-07-25:
See details here https://github.com/platformio/platformio-vscode-ide/issues/904#issuecomment-514157928 how to get errors

HenSpi, 2019-08-01:
I've done a "save as" from the Developer Tools Window. Hopefully that helps you for finding this error. 

console.ts:134 [Extension Host] runCommand platformio (7) ["-f", "-c", "vscode", "home", "--port", 8010, "--no-open"] {}
console.ts:134 [Extension Host] Error: 	at S.Promise (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)	at ChildProcess.h (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:134
$logExtensionHostMessage @ mainThreadConsole.ts:38
_doInvokeHandler @ rpcProtocol.ts:394
_invokeHandler @ rpcProtocol.ts:379
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:558
a @ ipc.net.ts:421
e @ ipc.net.ts:428
fire @ event.ts:558
_receiveMessage @ ipc.net.ts:715
S._socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:578
fire @ event.ts:558
acceptChunk @ ipc.net.ts:236
_register._socket.onData.e @ ipc.net.ts:197
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:134 [Extension Host] runCommand platformio (7) ["-f", "-c", "vscode", "home", "--port", 8010, "--no-open"] {}
console.ts:134 [Extension Host] Error: 	at S.Promise (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)	at ChildProcess.h (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:134
$logExtensionHostMessage @ mainThreadConsole.ts:38
_doInvokeHandler @ rpcProtocol.ts:394
_invokeHandler @ rpcProtocol.ts:379
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:558
a @ ipc.net.ts:421
e @ ipc.net.ts:428
fire @ event.ts:558
_receiveMessage @ ipc.net.ts:715
S._socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:578
fire @ event.ts:558
acceptChunk @ ipc.net.ts:236
_register._socket.onData.e @ ipc.net.ts:197
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:134 [Extension Host] runCommand platformio (7) ["-f", "-c", "vscode", "home", "--port", 8010, "--no-open"] {}
console.ts:134 [Extension Host] Error: 	at S.Promise (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)	at ChildProcess.h (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:134
$logExtensionHostMessage @ mainThreadConsole.ts:38
_doInvokeHandler @ rpcProtocol.ts:394
_invokeHandler @ rpcProtocol.ts:379
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:558
a @ ipc.net.ts:421
e @ ipc.net.ts:428
fire @ event.ts:558
_receiveMessage @ ipc.net.ts:715
S._socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:578
fire @ event.ts:558
acceptChunk @ ipc.net.ts:236
_register._socket.onData.e @ ipc.net.ts:197
t @ ipc.net.ts:28
emit @ events.js:182
[... 31 lines cut ...]
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:134 [Extension Host] Error: 	at S.Promise (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)	at ChildProcess.h (/Users/Hendrik/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:134
$logExtensionHostMessage @ mainThreadConsole.ts:38
_doInvokeHandler @ rpcProtocol.ts:394
_invokeHandler @ rpcProtocol.ts:379
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:558
a @ ipc.net.ts:421
e @ ipc.net.ts:428
fire @ event.ts:558
_receiveMessage @ ipc.net.ts:715
S._socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:578
fire @ event.ts:558
acceptChunk @ ipc.net.ts:236
_register._socket.onData.e @ ipc.net.ts:197
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94

HenSpi, 2019-08-22:
Hey everybody,
somebody in different internet forums found a way to fix that error. Here is a small description what I did and where I found that informations:

At first I need to install Python pip back the MacOS Catalina. This was described here: https://www.shellhacks.com/python-install-pip-mac-ubuntu-centos/

After that step I was able to install the Python setuptools. This was described here:
https://community.platformio.org/t/platformio-error-on-mac/5356/7

And at least and very important was to delete the ~/.platformio/penv folder as described in the ink above.

After restarting the Visual Studio Code again the PlatformIO functions are working again.

Thanks to all who find a solution for this issue!!!

Regards
  Hendrik

HenSpi, 2019-08-22:
Because of finding a solution for that Issue I close this issue.
Thanks to all the developers for that great PlatformIO-Tool!

closed 2019-08-22 by HenSpi