platformio/platformio-vscode-ide #905: Installation Manager

vparonov, 2019-07-17:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.36.0
PIO IDE: v1.8.0
System: Darwin, 18.2.0, x64

# Exception
```
Error: PIP Core: /Users/vangel/.platformio/penv/bin/python: No module named pip

	at f.value.o.runCommand (/Users/vangel/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:29748)
	at ChildProcess.h (/Users/vangel/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-07-17:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

vparonov, 2019-07-17:
> echo $PATH
/Users/vangel/.cabal/bin:/Users/vangel/.ghcup/bin:/Users/vangel/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/go/bin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands
 
> echo $PYTHONPATH

> python --version
Python 2.7.15
 
> pip --version
pip 10.0.1 from /usr/local/lib/python2.7/site-packages/pip (python 2.7)

> pip  list

Package                       Version
----------------------------- -------
astroid                       1.6.2
backports.functools-lru-cache 1.5
configparser                  3.5.0
enum34                        1.1.6
futures                       3.2.0
isort                         4.3.4
lazy-object-proxy             1.3.1
mccabe                        0.6.1
pip                           10.0.1
protobuf                      3.5.1
pylint                        1.8.3
setuptools                    39.0.1
singledispatch                3.4.0.3
six                           1.11.0
wheel                         0.31.0
wrapt                         1.10.11
You are using pip version 10.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Update: I restarted VSCode and the Platform IO page opened without errors …… I don’t know if it’s a sign that everything is ok

ivankravets, 2019-07-17:
It is good that you resolved it. I mark this issue as a duplicate of https://github.com/platformio/platformio-vscode-ide/issues/815

We will work on it.

closed 2019-07-17 by ivankravets