platformio/platformio-vscode-ide #910: Installation Manager

HAJM, 2019-07-18:
%23 Description of problem

Installing PlatformIO Core...
Please do not close this window and do not open other folders until this process is completed.
Failed to install PlatformIO IDE.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: PIP Core: platformio requires Python '>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*' but the running Python is 3.4.0
You are using pip version 18.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

	at f.value.o.runCommand (/Users/HAJM/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:29748)
	at ChildProcess.h (/Users/HAJM/.vscode/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

HAJM, 2019-07-18:
I have upgraded pip to version 19.1.1, but I still get the same error message, namely that I am using pip version 18.1.  I don't know how to downgrade Python to a version >= 2.7 and < 3.0.  More generally, I don't really understand what makes Visual Studio Code and PlatformIO tick, or what pip or Python do in this context, so please make any suggestions simple  and complete.  Thanks.

ivankravets, 2019-07-19:
> platformio requires Python '>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*' but the running Python is 3.4.0

Please uninstall Python 3.4.0 and install the latest 3.7

HAJM, 2019-07-19:
> 
> platformio requires Python '>=2.7, !=3.0., !=3.1., !=3.2., !=3.3., !=3.4.*' but the running Python is 3.4.0
> 
> Please uninstall Python 3.4.0 and install the latest 3.7

Thanks for your help.

I have removed Python 3.4 using the instructions for removing python 2.7 here 

https://stackoverflow.com/questions/3819449/how-to-uninstall-python-2-7-on-a-mac-os-x-10-6-4 <https://stackoverflow.com/questions/3819449/how-to-uninstall-python-2-7-on-a-mac-os-x-10-6-4>

by replacing 2.7 by 3.4 in the commands. I then tried to install Python 3.7 using the instructions here

install-python3-mac <https://wsvincent.com/install-python3-mac/>

but had some problems. I already had homebrew installed, but checked it using ‘brew doctor’ and got the following error message

iMac:~ HAJM$ brew doctor
Error: Unrecognized Mach-O load command: 0x32

I didn’t know what that meant, or if it was important, but since I have used homebrew successfully to install other things, I went ahead and tried to install Python 3.7, but got the following errors

iMac:~ HAJM$ brew install python3
touch: /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/.git/FETCH_HEAD: Permission denied
touch: /usr/local/Homebrew/.git/FETCH_HEAD: Permission denied
fatal: Unable to create '/usr/local/Homebrew/.git/index.lock': Permission denied
error: could not lock config file .git/config: Permission denied
Cannot rebase: You have unstaged changes.
Please commit or stash them.
==> Installing dependencies for python: pkg-config, sphinx-doc, gdbm, makedepend, openssl, readline, sqlite, xz
Error: Cannot link pkg-config
Another version is already linked: /usr/local/Cellar/pkg-config/0.29.1_2

How can I fix this?

Is it just a problem with permissions or could it be that I have messed up, because when I first copied and pasted the command to remove the symbolic links for Python 3.4, I forgot to replace 2.7 by 3.4 in the command?  I corrected this, but I have removed all the 2.7 links. I hope this is not the problem as I don’t know how to reinstall the links.

Can you help me further

Thanks

Hugh McIntyre

ivankravets, 2019-07-19:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

HAJM, 2019-07-19:
> 
> Please provide an output from a system Terminal
> 
> echo $PATH
> echo $PYTHONPATH
> python --version
> pip --version
> pip list
Thanks for replying to me.  Here is the output:

iMac:~ HAJM$ echo $PATH
/Library/Frameworks/Python.framework/Versions/3.4/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
iMac:~ HAJM$ 
iMac:~ HAJM$ echo $PYTHONPATH

iMac:~ HAJM$ python --version
Python 2.7.10
iMac:~ HAJM$ 
iMac:~ HAJM$ pip --version
-bash: pip: command not found
iMac:~ HAJM$ 
iMac:~ HAJM$ pip list
-bash: pip: command not found
iMac:~ HAJM$ 

BTW, after I failed to install Python 3.7, I killed Visual Studio Code and restarted it and it installed the PlatformIO IDE update without a problem, using Python 2.7.10.  However, I would still like to be able to update Python to 3.7.

HAJM, 2019-07-20:
Further to my previous message, I have discovered that the folder

/Library/Frameworks/Python.framework/Versions/

is empty. No version 2.7 or 3.4.  Obviously this is my problem with the missing pio command, but how do I fix it?

HAJM, 2019-07-20:
Further to my previous two posts, here is the current position.

I found  a python.Framework version 2.7 in Systems/Library/Frameworks/Versions and copied it into /Library/Frameworks/Versions. It did not contain the pio command, but i found it elsewhere and copied that in as well.

To check if it all worked ok before trying to flash my Anet A6 board, I decided to try it out on a (Chinese clone) Arduino Uno, so I created a small PlatformIO project which I was able to upload to the Uno using it’s boot loader.

When I tried to flash it directly, using a USBasp programmer, it didn’t work. Here is the output I got.

iMac:Blink10Secs HAJM$ pio run -t program
Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR 1.14.0 > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.0.0, tool-avrdude 1.60300.190628 (6.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio/build/uno/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.4% (used 9 bytes from 2048 bytes)
PROGRAM: [          ]   3.0% (used 976 bytes from 32256 bytes)
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Use manually specified: /dev/cu.wchusbserial1410
Programming .pio/build/uno/firmware.hex
avrdude: ser_open(): can't open device "unknown": No such file or directory

avrdude done.  Thank you.

*** [program] Error 1
===================================================================================== [ERROR] Took 0.62 seconds 

The project’s platform.ino file is

;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port =ç
;upload_port = /dev/ttys000
;upload_port = /dev/ttys001

The /dev/cu.wchusbserial1410 port worked for the normal upload using the bootloader. TI tried the other two ports as they where listed as belonging to me in /dev.  The system list the uSBasp device when I look in About This Mac- System Report - Hardware - USB.

If I don’t specify any upload port in platform.ino, I get this error.

Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [program] Explicit exit, status 1
==============================

Any idea what I can do to specify the correct port

ivankravets, 2019-07-23:
Does it work now? Please check that you see your Uno board via `pio device list` command.

closed 2019-07-23 by ivankravets

HAJM, 2019-07-25:
Sorry not to have replied sooner because of holidays and family commitments. I have now successfully installed Marlin after uploading a new Optiboot bootloader to the board using the arduino IDE and then uploading Marlin from PlatformIO.

Thanks for all your help.