platformio/platformio-vscode-ide #913: Start PIO Home Server

tsoe007, 2019-07-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Windows_NT, 6.1.7601, ia32

%23 Exception
```
Error: Error: Could not find a version that satisfies the requirement '~2.34.190418' for your system 'windows_x86'

	at P.Promise (C:\Users\galactica001\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:15396)
	at ChildProcess.h (C:\Users\galactica001\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\galactica001\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

CmdrChaos, 2019-07-23:
same issue on Mac

ivankravets, 2019-07-23:
> Could not find a version that satisfies the requirement '~2.34.190418'

This is a very strange issue. I don't how PlatformIO Core installer allowed you to use obsolete Python 3.4. See 

* https://github.com/platformio/platformio-core/blob/develop/platformio/managers/core.py#L29
* https://github.com/platformio/platformio-node-helpers/blob/master/src/misc.js#L174

Or, maybe you have multiple Python interpreter in the system. Please remove them and install the latest Python 3.7.

closed 2019-07-23 by ivankravets