platformio/platformio-vscode-ide #914: Installation Manager

develooper1994, 2019-07-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Windows_NT, 6.1.7601, x64

%23 Exception
```
Error: Conda Virtualenv: 
RemoveError: 'setuptools' is a dependency of conda and cannot be removed from
conda's operating environment.

	at f.value.o.runCommand (C:\Users\mustafa.caglar\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:26889)
	at ChildProcess.h (C:\Users\mustafa.caglar\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.EventEmitter.emit (domain.js:442:20)
	at ChildProcess.cp.emit (C:\Users\mustafa.caglar\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-07-23:
> Error: Conda Virtualenv: 

You have broken Conda installation in the system. Please re-install or remove it from the system.

closed 2019-07-23 by ivankravets

QGB, 2022-04-23:
```
Microsoft Windows [版本 10.0.18363.1316]
(c) 2019 Microsoft Corporation。保留所有权利。
C:\Users\qgb>cd /test/esp32

C:\test\ESP32>
C:\test\ESP32>
C:\test\ESP32>python -m pioinstaller.__main__
C:\QGB\Anaconda3\python.exe: Error while finding module specification for 'pioinstaller.__main__' (ModuleNotFoundError: No module named 'pioinstaller')

C:\test\ESP32>cd get_platformio_DEPENDENCIES

C:\test\ESP32\get_platformio_DEPENDENCIES>python -m pioinstaller.__main__
['C:/test/ESP32/get_platformio_DEPENDENCIES', 'C:\\test\\ESP32\\get_platformio_DEPENDENCIES', 'C:\\QGB\\Anaconda3\\python37.zip', 'C:\\QGB\\Anaconda3\\DLLs', 'C:\\QGB\\Anaconda3\\lib', 'C:\\QGB\\Anaconda3', 'C:\\Users\\qgb\\AppData\\Roaming\\Python\\Python37\\site-packages', 'C:\\QGB\\Anaconda3\\lib\\site-packages', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\binwalk-2.2.1-py3.7.egg', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\py2many-0.2.1-py3.7.egg', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\win32', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\win32\\lib', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\Pythonwin']
Installer version: 1.1.1
Platform: Windows-10
Python version: 3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Python path: C:\QGB\Anaconda3\python.exe
Creating a virtual environment at C:\Users\qgb\.platformio\penv
> c:\test\esp32\get_platformio_dependencies\pioinstaller\python.py(196)find_compatible_pythons()
-> for item in candidates:
(Pdb) exit
Traceback (most recent call last):
  File "C:\QGB\Anaconda3\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\QGB\Anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\__main__.py", line 138, in <module>
    sys.exit(main())
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\__main__.py", line 134, in main
    return cli(obj={})  # pylint: disable=no-value-for-parameter, unexpected-keyword-arg
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\core.py", line 1637, in invoke
    super().invoke(ctx)
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:/test/ESP32/get_platformio_DEPENDENCIES\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\__main__.py", line 62, in cli
    core.install_platformio_core(shutdown_piohome, dev, ignore_python)
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\core.py", line 73, in install_platformio_core
    ignore_pythons=ignore_pythons,
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\core.py", line 97, in _install_platformio_core
    penv_dir = penv.create_core_penv(ignore_pythons=ignore_pythons)
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\penv.py", line 52, in create_core_penv
    for python_exe in python.find_compatible_pythons(ignore_pythons):
  File "C:\test\ESP32\get_platformio_DEPENDENCIES\pioinstaller\python.py", line 196, in find_compatible_pythons
    for item in candidates:
    for item in candidates:
  File "C:\QGB\Anaconda3\lib\bdb.py", line 88, in trace_dispatch
    return self.dispatch_line(frame)
  File "C:\QGB\Anaconda3\lib\bdb.py", line 113, in dispatch_line
    if self.quitting: raise BdbQuit
bdb.BdbQuit

C:\test\ESP32\get_platformio_DEPENDENCIES>python -m pioinstaller.__main__
['C:/test/ESP32/get_platformio_DEPENDENCIES', 'C:\\test\\ESP32\\get_platformio_DEPENDENCIES', 'C:\\QGB\\Anaconda3\\python37.zip', 'C:\\QGB\\Anaconda3\\DLLs', 'C:\\QGB\\Anaconda3\\lib', 'C:\\QGB\\Anaconda3', 'C:\\Users\\qgb\\AppData\\Roaming\\Python\\Python37\\site-packages', 'C:\\QGB\\Anaconda3\\lib\\site-packages', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\binwalk-2.2.1-py3.7.egg', 'C:\\QGB\\Anaconda3\\lib\\site-packages\\py2manlib\\site-packages\\Pythonwin']
Installer version: 1.1.1
Python version: 3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Python path: C:\QGB\Anaconda3\python.exe
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
> c:\test\esp32\get_platformio_dependencies\pioinstaller\core.py(99)_install_platformio_core()
-> python_exe = os.path.join(
(Pdb) l
[... 86 lines cut ...]
  Downloading sniffio-1.2.0-py3-none-any.whl (10 kB)
Collecting zipp>=0.5
  Using cached zipp-3.8.0-py3-none-any.whl (5.4 kB)
Collecting pyparsing!=3.0.5,>=2.0.2
  Using cached pyparsing-3.0.8-py3-none-any.whl (98 kB)
Building wheels for collected packages: platformio
  Building wheel for platformio (setup.py) ... done
  Created wheel for platformio: filename=platformio-5.2.5-py3-none-any.whl size=344464 sha256=465480c3308e9ea3b5e485b7b202258c545761600ae8a21989196b44732cc12d
  Stored in directory: c:\users\qgb\appdata\local\pip\cache\wheels\9c\82\ba\d7d0cfdba793796822036a1657d9bd10a9a6a03f98eba83bdb
Successfully built platformio
[... cut ...]
PlatformIO Core has been successfully installed into an isolated environment `C:\Users\qgb\.platformio\penv`!

The full path to `platformio.exe` is `C:\Users\qgb\.platformio\penv\Scripts\platformio.exe`

If you need an access to `platformio.exe` from other applications, please install Shell Commands
(add PlatformIO Core binary directory `C:\Users\qgb\.platformio\penv\Scripts` to the system environment PATH variable):

See https://docs.platformio.org/page/installation.html#install-shell-commands

C:\test\ESP32\get_platformio_DEPENDENCIES>
```

QGB, 2022-04-23:
pio 第一次 新建工程 ，疑似卡住
```
Please wait...
The first initialization requires Internet connection and may take a while (need to install dependent toolchains, frameworks, SDKs).
Please be patient and let the initialization complete.
```

ivankravets, 2022-04-23:
PlatformIO does not support Anaconda. Please remove anaconda and install the official Python interpreter. https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

QGB, 2022-04-24:
@ivankravets  

I delete `get_platformio\DEPENDENCIES\pioinstaller\python.py`

```
    if is_conda():
        raise exception.IncompatiblePythonError("Conda is not supported")
``` 
code, And force install in anaconda python :)

erikaemma, 2022-10-03:
@QGB I can't find those code

QGB, 2022-10-03:
> @QGB I can't find those code

it is a zip package