platformio/platformio-vscode-ide #918: `pio lib install` always ends in `Acces denied` OS error

Bascy, 2019-07-20:
When running `pio lib install`, the dependent library is cloned from github into a temp folder in `.pio\libdeps` but then the process ends with the error pasted at the bottom of this issue.
I've tried running VSCode with administrator rights, but the same problem occurs.

Using PlatformIO 4.0.1b3 
Python 3.7.4

>  [Error 5] Access is denied: 'D:\\PlatformIO\\Projects\\csWaterflowShuntNode\\.pio\\libdeps\\debug\\_tmp_installing-f51hte-package' 
> Please manually remove the file `D:\PlatformIO\Projects\csWaterflowShuntNode\.pio\libdeps\debug\_tmp_installing-f51hte-package`
> Error: Traceback (most recent call last):
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
>     cli(None, None, None)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
>     return self.main(*args, **kwargs)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
>     rv = self.invoke(ctx)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
>     return super(PlatformioCLI, self).invoke(ctx)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
>     return callback(*args, **kwargs)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\click\decorators.py", line 17, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 158, in lib_install
>     force=force)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 382, in install
>     force=force)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 353, in install
>     force=force)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 724, in install
>     track=True)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 523, in _install_from_url
>     return self._install_from_tmp_dir(_tmp_dir, requirements)
>   File "c:\users\bas\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 590, in _install_from_tmp_dir
>     shutil.move(pkg_dir, join(self.package_dir, target_dirname))
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 300, in move
>     rmtree(src)
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 247, in rmtree
>     rmtree(fullname, ignore_errors, onerror)
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 247, in rmtree
>     rmtree(fullname, ignore_errors, onerror)
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 247, in rmtree
>     rmtree(fullname, ignore_errors, onerror)
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 252, in rmtree
>     onerror(os.remove, fullname, sys.exc_info())
>   File "C:\Users\Bas\.platformio\python27\Lib\shutil.py", line 250, in rmtree
>     os.remove(fullname)
> WindowsError: [Error 5] Access is denied: 'D:\\PlatformIO\\Projects\\csWaterflowShuntNode\\.pio\\libdeps\\debug\\StandardCplusplus\\.git\\objects\\02\\8e86f36b5c478c05fe5cd4165267a7bd1b2fba'

ivankravets, 2019-08-31:
It seems like antivirus tools block PlatformIO

closed 2019-08-31 by ivankravets

Bmooij, 2019-10-08:
I may have found the problem.
From the release notes from the `Git for Windows v2.23.0 Release Notes`
>Known issues
On Windows 10 before 1703, or when Developer Mode is turned off, special permissions are required when cloning repositories with symbolic links, therefore support for symbolic links is disabled by default. Use git clone -c core.symlinks=true <URL> to enable it, see details [here](https://github.com/git-for-windows/git/wiki/Symbolic-Links).

I've turned on the developer settings in windows 10, and rebooted.
Now the permission errors are gone.

Bmooij, 2019-10-08:
I may found the problem. It happens in the following function: [shutil.move(tmp_dir, pkg_dir)](https://github.com/platformio/platformio-core/blob/eda02750ae489d6c4f7b978c98ac213da75aca03/platformio/managers/package.py#L585)

This is because the files in `.git\objects\xx\xxxxxxxxxxxxxxxxxx` are readonly