platformio/platformio-vscode-ide #920: Debugger: Data in assembly view isn't loaded

mirao, 2019-07-21:
It's a regression in PIO IDE 1.8.0.
It works well with 1.7.1 (fortunately VSCode allows upgrade of extension to older versions)

**Steps to reproduce:**
- Open a project and debug (F5)
- DISASSEMBLY => Switch to assembly view

**Expected result:**

- you get a window (`main.dbgasm`) with assembler instructions

**Actual result:**

- you get an empty window with loading indicator. Instructions are never loaded:

[image]

**Used SW:**

- VSCode 1.36.1
- Ubuntu 19.0.4
- PlatformIO IDE extension 1.8.0
- pio 4.0.0

**Used HW:**

- STM32F407-DISC1

ivankravets, 2019-07-23:
Please update to https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.1

We would be thankful if you update your review => https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide&ssr=false#review-details

This rating helps us to get better adoption of PlatformIO in the embedded systems industry. Thanks!

closed 2019-07-23 by ivankravets

mirao, 2019-07-23:
I upgraded to 1.8.1 and it's working again, thanks for your effort.
Review updated.

commit dfd3a528 (2019-07-23): Fixed an issue with broken Code Disassembly feature of PIO Unified Debugger // Resolve #920