platformio/platformio-vscode-ide #921: Minor: Devices in each environment

pfeerick, 2019-07-22:
An entry for 'devices' is present in each environment, which fails as it appends invalid syntax (`--environment` + the environment name) to the `device list` command.

[image]

```
> Executing task in folder tracerWifiLCD: C:\Users\Peter\.platformio\penv\Scripts\platformio.exe device list --environment d1mini-ota <

Usage: platformio device list [OPTIONS]  
Try "platformio device list -h" for help.

Error: no such option: --environment

Terminal will be reused by tasks, press any key to close it.
```

ivankravets, 2019-07-23:
Plese update to https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.1

pfeerick, 2019-07-23:
Looks good. Thanks! :)

closed 2019-07-23 by pfeerick