platformio/platformio-vscode-ide #922: ./vscode/extensions.json constantly being overwritten by PIO

voney, 2019-07-22:
After installing the platformio extension it overwrites the `extension.json` file with the below:
```json
{
	// See http://go.microsoft.com/fwlink/?LinkId=827846
	// for the documentation about the extensions.json format
	"recommendations": [
		"platformio.platformio-ide"
	]
}
```
It continues to do this every time the workspace is opened.
It should instead check the contents of the file and insert the recommendation if it's not present.

voney, 2019-09-08:
I'm not sure why this has been labelled an enhancement? This extension clobbers workspace settings, that's a bug.

pfeerick, 2019-10-20:
@voney FYI, it's because what you've asked for is an enhancement to part of the PlatformIO behaviour... I believe it alters `extension.json` for a reason... so that VSCode will prompt for the PlatformIO to be installed if you open the project on a machine without PlatformIO... but obviously it would be good if it did that without clobbering any user added extension customisations... hence 'enhancement'. ;)

thawkins, 2019-10-21:
How do we stop platformio from being installed on a project that is not a platformio project, I use vcode from lots of other tasks, such as java and web development, and its a pain to have it attempt install itself every-time i switch to a new folder.  Cant we just have a manual "set this as a platformio" button and ditch all this automated behavior, or at least ditch it if there is no platfoirmio.ini file in the project folder.

pfeerick, 2019-10-21:
When you say ...

> to have it attempt install itself every-time i switch to a new folder

... do you mean when it says something along the lines of 'Checking PlatformIO Core Installation' when you open a non-platformIO project?

If you don't want PlatformIO to be running at all, you can turn extensions on and off by workspace... thus you can have a C++ optimised workspace vs a Web workspace vs a java workspace, etc.

pfeerick, 2019-11-28:
I think another aspect to this is that if the user switches between Linux and Windows, for example, the line endings are changed from CRLF to LF, which then gets tracked with git. Doing the 'skip change to file if recommendation present' enhancement should fix that problem as well.

voney, 2020-06-30:
I Just tested this with the latest platformio (4.3.4) and this no longer occurs. Not sure when it was fixed but now it only adds itself to the file instead of clobbering the whole thing.

closed 2020-06-30 by voney

Filiprogrammer, 2021-08-21:
It still changes the line endings to CRLF every time you start VSCode.

forkineye, 2021-09-26:
> It still changes the line endings to CRLF every time you start VSCode.

I'm experiencing this as well on MacOS. Doesn't seem to do it on Windows.