platformio/platformio-vscode-ide #923: Installation Manager

ButchAnton, 2019-07-22:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.0-insider
PIO IDE: v1.8.0
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: /usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/__init__.py:15: UserWarning: The virtualenv distutils package at %s appears to be in the same location as the system distutils%3F
  warnings.warn("The virtualenv distutils package at %s appears to be in the same location as the system distutils%3F")
Traceback (most recent call last):
  File "/Users/I831533/.platformio/.cache/tmp-44075xpi6kuqZ9hdZ/virtualenv-16.1.0/src/virtualenv.py", line 2462, in <module>
    main()
  File "/Users/I831533/.platformio/.cache/tmp-44075xpi6kuqZ9hdZ/virtualenv-16.1.0/src/virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "/Users/I831533/.platformio/.cache/tmp-44075xpi6kuqZ9hdZ/virtualenv-16.1.0/src/virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "/Users/I831533/.platformio/.cache/tmp-44075xpi6kuqZ9hdZ/virtualenv-16.1.0/src/virtualenv.py", line 1198, in install_python
    fix_lib64(lib_dir, symlink)
  File "/Users/I831533/.platformio/.cache/tmp-44075xpi6kuqZ9hdZ/virtualenv-16.1.0/src/virtualenv.py", line 1620, in fix_lib64
    if not [p for p in distutils.sysconfig.get_config_vars().values() if isinstance(p, basestring) and "lib64" in p]:
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/__init__.py", line 105, in sysconfig_get_config_vars
    real_vars = old_get_config_vars(*args)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/sysconfig.py", line 511, in get_config_vars
    srcdir = os.path.dirname(get_makefile_filename())
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/sysconfig.py", line 258, in get_makefile_filename
    lib_dir = get_python_lib(plat_specific=0, standard_lib=1)
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/__init__.py", line 94, in sysconfig_get_python_lib
    prefix = sys.real_prefix
AttributeError: module 'sys' has no attribute 'real_prefix'

Using base prefix '/usr/local/bin/../Cellar/python/3.7.4/bin/../Frameworks/Python.framework/Versions/3.7'

    at t.value (/Users/I831533/.vscode-insiders/extensions/platformio.platformio-ide-1.8.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/installer/stages/platformio-core.js:271:21)
```

ivankravets, 2019-07-23:
```bash
$ which python3
/usr/local/bin/python3

$ readlink /usr/local/bin/python3
../Cellar/python/3.7.4/bin/python3

$ python3 -m venv /tmp/tdenv
$ source /tmp/tdenv/bin/activate
(tdenv) ~ $ pip list
Package    Version
---------- -------
pip        19.0.3
setuptools 40.8.0
```

It seems that your Homebrew's Python is broken. Please re-install it.

LongLiveCHIEF, 2019-07-24:
Looks like we're getting a lot of this in the last 2 days. So far #927 #928 #930 are all duplicates. I've noticed that it appears to be affecting all platforms, as the report logs from the issues mentioned above include Mac, Windows, and standard Linux respectively.

Ironically, I am running WSL Remote on WSL2, and was going to report this figuring it would be limited to this unique setup.  So my logs are attached below :smiley:

---

## Configuration

VSCode: 1.36.1
PIO IDE: v1.8.1
System: Linux, 4.19.43-microsoft-standard, x64

## Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/chief/.platformio/.cache/tmp-57037txfZ4GrLeiJ/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/chief/.vscode-server/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:31066)
	at process._tickCallback (internal/process/next_tick.js:68:7)
```

ivankravets, 2019-07-24:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

ButchAnton, 2019-07-24:
`
$ echo $PATH

/usr/local/share/python:/usr/local/git/bin:/usr/local/bin:/usr/local/sbin:/usr/local/mysql/bin:/usr/bin/X11:/usr/local/bin/X11:/Users/I831533/bin:/opt/local/bin:/opt/xindice/bin:/opt/local/lib/postgresql84/bin:/usr/bin:/bin:/usr/contrib/bin:/sbin:/usr/sbin:/usr/lib:/usr/ucb:/usr/etc:/etc:~/.pyenv/bin/:/usr/local/opt/go/libexec/bin:/Users/butch/Library/Python/3.7/bin:/usr/local/share/python:/usr/local/git/bin:/usr/local/bin:/usr/local/sbin:/usr/local/mysql/bin:/usr/bin/X11:/usr/local/bin/X11:/Users/I831533/bin:/opt/local/bin:/opt/xindice/bin:/opt/local/lib/postgresql84/bin:/usr/bin:/bin:/usr/contrib/bin:/sbin:/usr/sbin:/usr/lib:/usr/ucb:/usr/etc:/etc:~/.pyenv/bin/:/usr/local/opt/go/libexec/bin:/Users/butch/Library/Python/3.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:.:/Users/I831533/.platformio/penv/bin:/Users/I831533/src/esp_toolchain/xtensa-esp32-elf/bin:/usr/local/heroku/bin:/Users/I831533/.rvm/bin:.:/Users/I831533/.platformio/penv/bin:/Users/I831533/src/esp_toolchain/xtensa-esp32-elf/bin:/usr/local/heroku/bin:/Users/I831533/.rvm/bin

$ echo $PYTHONPATH

$ python --version
Python 3.7.4

$ pip --version
pip 8.1.1 from /Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg (python 2.7)

$ pip list
altgraph (0.10.2)
backports.ssl-match-hostname (3.5.0.1)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
paho-mqtt (1.1)
pip (8.1.1)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (3.1.1)
python-dateutil (1.5)
pytz (2013.7)
requests (2.11.0)
scipy (0.13.0b1)
setuptools (18.5)
six (1.4.1)
socketIO-client (0.7.0)
vboxapi (1.0)
websocket-client (0.37.0)
xattr (0.6.4)
You are using pip version 8.1.1, however version 19.2.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

$ pip3 --version
/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/distutils/__init__.py:15: UserWarning: The virtualenv distutils package at %s appears to be in the same location as the system distutils?
  warnings.warn("The virtualenv distutils package at %s appears to be in the same location as the system distutils?")
pip 19.1.1 from /usr/local/lib/python3.7/site-packages/pip (python 3.7)

$ pip install --upgrade pip
Collecting pip
  Using cached https://files.pythonhosted.org/packages/62/ca/94d32a6516ed197a491d17d46595ce58a83cbb2fca280414e57cd86b84dc/pip-19.2.1-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Uninstalling pip-8.1.1:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/commands/install.py", line 317, in run
    prefix=options.prefix_path,
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/req/req_set.py", line 726, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/req/req_install.py", line 746, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 300, in move
    rmtree(src)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 250, in rmtree
    os.remove(fullname)
OSError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/pip-8.1.1-py2.7.egg/EGG-INFO/PKG-INFO'
You are using pip version 8.1.1, however version 19.2.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
`

Looks like there is an issue with pip (python is version 3, pip is version 2, pip3 seems FUBAR).

LongLiveCHIEF, 2019-07-24:
@ivankravets just saw this as I came back to record my fix. Your hunch was correct.

I started by running `python --version` which told me python was not available, but `python3` _was_ available.

So, from here, my process was this:
1. Ensure latest `pyton3` was installed
2. alias `python=python3`
3. install `python-pip3`
4. Install `virtualenv`
5. Activate pip venv 
6. use pip to install platformio
7. restart all the things
8. Success!

```
rm -rf ~/.platformio/penv
sudo apt-get update
sudo apt-get install python3 python3-pip direnv
echo eval \"\$(direnv hook bash\)\" >> ~/.bashrc
```
closed and opened new shell, then:
```
python -m pip install virtualenv
```
close/open shell, then:
```
virtualenv -p python3 ~/.platformio/penv
```
 close/open shell then: created a file at `$HOME/.envrc` with the content:

```
if (tty -s); then
  source $HOME/.platformio/penv/bin/activate
fi
``
Now restart VS Code, and error is resolved!

ivankravets, 2019-07-25:
Python 3 has built in virtual env. You can use it via `python -m venv -h`

LongLiveCHIEF, 2019-07-26:
> Python 3 has built in virtual env.

Good to know! I had no idea! Thanks!

ivankravets, 2019-08-11:
@ButchAnton could we close this issue?

closed 2019-08-12 by ButchAnton

matt-uib, 2025-07-23:
I had this error in vs code in debugpy while debugging a plotly dash app and the only reason was that i at some time in the past enabled breaking on "Raised Exceptions" in vs code. So the exception was shown, but there was no real problem, as it was handled. Maybe someone had the same problem...

[image]