platformio/platformio-vscode-ide #928: Installation Manager

tkykst1968, 2019-07-24:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.0
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 C:\msys32\home\.platformio\python27\python.exe: No module named pip

	at f.value.o.runCommand (C:\Users\satou\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:29748)
	at ChildProcess.h (C:\Users\satou\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\platformio-node-helpers\dist\index.js:1:5839)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\satou\.vscode\extensions\platformio.platformio-ide-1.8.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

LongLiveCHIEF, 2019-07-24:
recommend closing this as duplicate and merging with #923

tkykst1968, 2019-07-25:
I'm sorry to duplicate you.

LongLiveCHIEF, 2019-07-25:
You didn't.

closed 2019-07-26 by tkykst1968