platformio/platformio-vscode-ide #930: Installation Manager

coralnode, 2019-07-24:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.1
System: Linux, 4.18.0-25-generic, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/rouzbeh/.platformio/.cache/tmp-2503HfhGUBXpNunf/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/rouzbeh/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:31159)
```

LongLiveCHIEF, 2019-07-24:
recommend closing this as duplicate and merging with #923

ivankravets, 2019-08-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/907

closed 2019-08-11 by ivankravets