platformio/platformio-vscode-ide #933: Task error

HamzaHajeir, 2019-07-26:
Hi

I am getting these errors whenever i build a newly created project : 

`Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Build",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Build",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.`

I've deleted tasks.json , nothing changed
I've recreated tasks.json and filled it according http://docs.platformio.org/en/latest/ide/vscode.html#task-runner

`{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "task": "Monitor",
            "problemMatcher": [
                "$platformio"
            ],
            "presentation": {
                "panel": "new"
            }
        }
    ]
}`

Nothing changed, I still receive the same error.

my platformio.ini : 

`[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps =
  #ESPArto
  5973
  #Pubsubclient
  89
`

And for an empty sketch of Arduino UNO, the same thing happens with these messages : 

`Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Build",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ],
    "presentation": {
        "panel": "new"
    }
}
The task will be ignored.`

platformio.ino : 
`[env:uno]
platform = atmelavr
board = uno
framework = arduino
`

Any help .. ?

ivankravets, 2019-08-11:
1. Please remove `tasks.json` from a project.
2. Restart VSCode.

It should work now.

closed 2019-08-11 by ivankravets

HamzaHajeir, 2019-08-12:
Nice, error disappeared for 'build', but it still appears for 'upload' command (ctrl + alt + u).

> Error: The PlatformIO task detection didn't contribute a task for the following configuration:
> {
>     "type": "PlatformIO",
>     "task": "Monitor",
>     "problemMatcher": [
>         "$platformio"
>     ],
>     "presentation": {
>         "panel": "new"
>     }
> }
> The task will be ignored.
> Error: The PlatformIO task detection didn't contribute a task for the following configuration:
> {
>     "type": "PlatformIO",
>     "task": "Monitor",
>     "problemMatcher": [
>         "$platformio"
>     ],
>     "presentation": {
>         "panel": "new"
>     }
> }
> The task will be ignored.

To understand better, does this error affect anything?

ivankravets, 2019-08-12:
Do you still have `tasks.json` in `.vscode` folder?

HamzaHajeir, 2019-08-12:
It just disappeared right now. 

I guess it's because I triggered this command (in trying to get another project work).
`platformio upgrade --dev`

Thanks @ivankravets