platformio/platformio-vscode-ide #938: Start PIO Home Server

deasonsv, 2019-07-31:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.1
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/deasonsv/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)
	at ChildProcess.h (/Users/deasonsv/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-08-11:
Could you open PlatformIO IDE Terminal and type:
```
pio home
```

Do you see any errors?

deasonsv, 2019-08-12:
MacBook-Sergey-V:~ deasonsv$ pio home
bash: pio: command not found

ivankravets, 2019-08-12:
You need to use VSCode > PlatformIO (left sidebar icon) > Quick Links > PlatformIO Core CLI.

deasonsv, 2019-08-12:
MacBook-Sergey-V:~ deasonsv$ pio home
Abort trap: 6
MacBook-Sergey-V:~ deasonsv$

ivankravets, 2019-08-12:
What is your `platformio --version`?

ivankravets, 2019-08-12:
Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

deasonsv, 2019-08-12:
deasonsv@MacBook-Sergey-V ~ % echo $PATH           
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Library/Apple/bin
deasonsv@MacBook-Sergey-V ~ % python --version     
Python 2.7.16
deasonsv@MacBook-Sergey-V ~ % pip --version   
pip 19.2.2 from /Library/Python/2.7/site-packages/pip-19.2.2-py2.7.egg/pip (python 2.7)
deasonsv@MacBook-Sergey-V ~ % pip list        
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package                                Version 
-------------------------------------- --------
altgraph                               0.10.2  
asn1crypto                             0.24.0  
bdist-mpkg                             0.5.0   
bonjour-py                             0.3     
cffi                                   1.12.2  
cryptography                           2.6.1   
enum34                                 1.1.6   
ipaddress                              1.0.22  
macholib                               1.5.1   
matplotlib                             1.3.1   
modulegraph                            0.10.4  
numpy                                  1.8.0rc1
pip                                    19.2.2  
py2app                                 0.7.3   
pycparser                              2.19    
pyobjc-core                            2.5.1   
pyobjc-framework-Accounts              2.5.1   
pyobjc-framework-AddressBook           2.5.1   
pyobjc-framework-AppleScriptKit        2.5.1   
pyobjc-framework-AppleScriptObjC       2.5.1   
pyobjc-framework-Automator             2.5.1   
pyobjc-framework-CFNetwork             2.5.1   
pyobjc-framework-Cocoa                 2.5.1   
pyobjc-framework-Collaboration         2.5.1   
pyobjc-framework-CoreData              2.5.1   
pyobjc-framework-CoreLocation          2.5.1   
pyobjc-framework-CoreText              2.5.1   
pyobjc-framework-DictionaryServices    2.5.1   
pyobjc-framework-EventKit              2.5.1   
pyobjc-framework-ExceptionHandling     2.5.1   
pyobjc-framework-FSEvents              2.5.1   
pyobjc-framework-InputMethodKit        2.5.1   
pyobjc-framework-InstallerPlugins      2.5.1   
pyobjc-framework-InstantMessage        2.5.1   
pyobjc-framework-LatentSemanticMapping 2.5.1   
pyobjc-framework-LaunchServices        2.5.1   
pyobjc-framework-Message               2.5.1   
pyobjc-framework-OpenDirectory         2.5.1   
pyobjc-framework-PreferencePanes       2.5.1   
pyobjc-framework-PubSub                2.5.1   
pyobjc-framework-QTKit                 2.5.1   
pyobjc-framework-Quartz                2.5.1   
pyobjc-framework-ScreenSaver           2.5.1   
pyobjc-framework-ScriptingBridge       2.5.1   
pyobjc-framework-SearchKit             2.5.1   
pyobjc-framework-ServiceManagement     2.5.1   
pyobjc-framework-Social                2.5.1   
pyobjc-framework-SyncServices          2.5.1   
pyobjc-framework-SystemConfiguration   2.5.1   
pyobjc-framework-WebKit                2.5.1   
pyOpenSSL                              19.0.0  
pyparsing                              2.0.1   
python-dateutil                        1.5     
pytz                                   2013.7  
scipy                                  0.13.0b1
setuptools                             41.0.1  
six                                    1.12.0  
wheel                                  0.33.1  
xattr                                  0.6.4   
deasonsv@MacBook-Sergey-V ~ %

deasonsv, 2019-08-12:
latest

ivankravets, 2019-08-15:
Could you remove `/Users/deasonsv/.platformio/penv` folder and restart VSCOde?

deasonsv, 2019-08-15:
Platformio reinstalls after remove, but the problem remains after restarting

ivankravets, 2019-08-15:
How did you install this `Python 2.7.16`? Via `brew`?

deasonsv, 2019-08-15:
No
from package
https://www.python.org/downloads/release/python-2716/

deasonsv, 2019-08-15:
> 15 авг. 2019 г., в 14:26, Sergey Vinnikov <deasonsv@gmail.com> написал(а):
>
> No
> from package
> https://www.python.org/downloads/release/python-2716/ <https://www.python.org/downloads/release/python-2716/>

wow& I’m reinstall pythone and remove /Users/deasonsv/.platformio/penv folder 
Open VSD without errors

Thanks

ivankravets, 2019-08-15:
That is why I asked. It was broken Python. 

Happy coding with PlatformIO!

closed 2019-08-15 by ivankravets