platformio/platformio-vscode-ide #942: PlatformIO IDE extension does not work with Remote Development using SSH

pbolduc, 2019-07-31:
I am trying to use Remote Development using SSH with PlatformIO IDE.  Remote Development using SSH is a preview feature, but would be great if this would work in the future.

https://code.visualstudio.com/docs/remote/ssh

I followed Scott Hanselman's blog to set up my environment, https://www.hanselman.com/blog/VisualStudioCodeRemoteDevelopmentOverSSHToARaspberryPiIsButter.aspx

My use case is I want to develop for the Raspberry PI using C/C++ and to use the awesome Platform IO to managed dependencies and build process.

PlatformIO IDE: 1.8.1

[image]

ivankravets, 2019-08-11:
Could you run in this RPi terminal `$ pio home --no-open`? Do you see any errors?

pbolduc, 2019-08-11:
No, there is no errors.  I get

```
  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

Open PIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.
```

[image]

pbolduc, 2019-08-12:
I had the window open since I took the screen shot above and the PIO Home screen and libraries show up...

ivankravets, 2019-08-13:
Ah, yes! It will not work. You run PIO Home Server on the remote machine via SSH where PIO Home Backend points to localhost. You can open PIO Home in your standalone browser via http://your-rpi-ip:8008.

IBlackI, 2019-12-29:
Is there any way to have the window in visual studio code point to a remote adress? This is also an issue when using Remote Developent through for example code-server

VishalDhayalan, 2020-01-09:
I'm having the same problem when trying to get it to work on the Raspberry Pi.
[image]
The PIO home page never loads and I get this error:
[image]

I also noticed the C/C++ extension that PIO depends on throws an error saying ARM is not supported:
[image]

Any fix for this?

piccaso, 2020-01-10:
As a workaround the connection issue adding this to the ssh_config seems to work:
```
LocalForward 127.0.0.1:8010 127.0.0.1:8010
```

However, when i created a new Project named 'bpTest' it was created as `~/Documents/PlatformIO/Projects\bpTest` where `Projects\bpTest` is a folder name.

Also C/C++ problem highlighting does not work.

So there is more to it than setting up port forwardung...

VishalDhayalan, 2020-01-18:
> As a workaround for the connection issue adding this to the ssh_config seems to work:
> 
> ```
> LocalForward 127.0.0.1:8010 127.0.0.1:8010
> ```
> 
> However, when i created a new Project named 'bpTest' it was created as `~/Documents/PlatformIO/Projects\bpTest` where `Projects\bpTest` is a folder name.
> 
> Also C/C++ problem highlighting does not work.
> 
> So there is more to it than setting up port forwardung...
> Edit:
> After fixing the path manually C/C++ problem highlighting works and I cant find any other issue.

I didn't have any ssh problems, but could you please elaborate on how you fixed the path manually? And is this on an ARM architecture (such as the raspberry pi)? Thanks.

Turiok, 2020-02-24:
Thx @VishalDhayalan.

Works for me :).
In your ssh/config file, The description of your remote, you add the line from @VishalDhayalan : 
LocalForward 127.0.0.1:8010 127.0.0.1:8010
It'll forward the port between your host and remote. So VScode will displayed on localhost:8010 the localhost:8010 of your VM.

Turiok, 2020-03-03:
Sadly, the window appear.
But when I want to import a arduino project.
I can't display my folder tree than one level.

```bash
$ ls -l workspace/
drwxrwxr-x  7 vagrant vagrant 4096 Feb  6 19:20 ESP32
drwxrwxr-x  9 vagrant vagrant 4096 Feb 22 12:28 MQTTnet
```
[image]

Even If I want to see the content of folder in /tmp,  doesn't work

Turiok, 2020-03-05:
Hello again, 

I think I found the problem :) 
So a summarize.
I run vscode on windows host with remote ssh extension on a linux guest.
platformio extension is installed on guest.
After add the port forwarding to my ssh config.
(To make it work, we can launch pio with --host 0.0.0.0 to accept outbound config)
I have the page but I can't navigate in my file system.

For me it's a problem of Windows/Linux useragent problem.
What are my tests.
I launch a new instance of pio on my guest 
```bash
pio home --host 0.0.0.0 --port 8008 --no-open
```

I launch firefox on my host with guest ip. Same problem than vscode.
But if I launch firefox on an other linux computer. Navigation works and folder creation too. Miracle :) 
I tried to install an extension on firefox to change useragent.
The goal is to send an useragent with linux OS with my firefox on windows. But this extension seems to not work for the websocket. on pio.
So I don't found a workaround for now.
If someone can help me. And I don't know the impact with vs code.

MBetters, 2020-06-02:
Another issue in remote-SSH is that PIO Home --> New Project is showing host filesystem, not guest filesystem.

septatrix, 2020-10-08:
> However, when i created a new Project named 'bpTest' it was created as `~/Documents/PlatformIO/Projects\bpTest` where `Projects\bpTest` is a folder name.

I do not exactly know what causes this. After looking a bit in the source code I was however able to find some places where we use the native `fs` module instead of the one provided by the VSCode API. This _may_ lead to such problems. Even if it does not lead to problems the usage of the native module is discouraged anyways as it apparently runs in the editor process unlike most other stuff accessible over the vscode api. Currently I do not have a remote machine setup with which I could test this but I may be able to in a few days...

paloj, 2021-03-17:
> [image]

I have this same issue. Im using vscode locally on windows with ssh remote connection to mac mini m1 which has an arduino board attached to it.

hazza64, 2021-03-21:
I am also having this exact same issue, even though from memory it was working a while ago.
Setup:

- VSCode installed on a Windows 10 desktop
- Remote - SSH (Nightly) extension installed locally
- PlatformIO IDE installed remotely on target (Raspberry Pi 4 running Rasbian)
- C/C++ Installed remotely (Raspberry Pi 4)

Opening the PIO Home screen shows nothing.

I was doing some research (but lost the links -_-) that suggests it may be an issue with it using the Windows Filesystem on the remote target (in this case Rasbian GNU/Linux). 
Seems like they're on it; issue linked here https://github.com/platformio/platformio-core/issues/3681

arpaterson, 2022-10-04:
I have also run into this.
Windows 11 VSCode+PlatformIO -> Remote SSH extension -> Jetson Orin developer kit -> Feather M4 Express
Platform IO extension installed thru client VScode on both client (win11) and orin (ubuntu 20.04)

I'd like to be able to use either of two work flows:
 - Code and build on windows 11, but remote upload to the Feather. (Only the upload tool runs remotely)
 - Code on windows 11 thru Remote SSH - so the project resides on the remote (PlatformIO itself, project files, and build tool all remote).

Hsun1031, 2023-02-22:
1. Use VS Code's SSH remote feature to access your computer remotely.
2. Go to the PlatformIO Extensions settings and set the "PlatformIO-IDE: Pio Home Server Http Port" to a value such as 8080 in your user settings.
3. Set the same port in the remote settings for [ssh:[www.example.com](http://www.example.com/)].
4. To forward the port, press F1 and select "Ports: Focus on Ports View", then choose "Forward a Port" and add port 8080.
5. Finally, you will be able to use PIO Home.

FabioPlunser, 2023-03-27:
@Hsun1031 What do youu mean by Set the same port in the remote settings?

Brechtjeu, 2023-04-14:
@Hsun1031 are you also able to create a new project this way? I cannot. Idon't see the remote filesystem and when I create the project, I get a folder at an unusable path with only half and installation (esp32 esp-idf framework)

Hsun1031, 2023-04-15:
@Brechtjeu I have tested that Ubuntu can connect to Ubuntu, but Windows cannot connect to Ubuntu.

luxk3, 2023-07-09:
Hello everybody, I was trying to solve a different issue, and I took a similar road to the one described here. I know my use-case is a bit different, but I wrote a short guide. I leave it here as reference, it might be helpful to somebody else.

[https://github.com/luxk3/ubuntu_server_remote_dev_platformio](https://github.com/luxk3/ubuntu_server_remote_dev_platformio)

luxk3

d330447535, 2024-06-14:
[https://code.visualstudio.com/api/advanced-topics/remote-extensions#forwarding-localhost](url)

d330447535, 2024-06-14:
Use the following code to solve the problem of not opening the page under SSH，View the api [asExternal](Urihttps://code.visualstudio.com/api/advanced-topics/remote-extensions#forwarding-localhost)
`    const pageUrl = await vscode.env.asExternalUri(
      vscode.Uri.parse(pioNodeHelpers.home.getFrontendUrl({
        start: startUrl,
        theme,
        workspace: extension.getEnterpriseSetting('defaultPIOHomeWorkspace'),
      }))
    );`

d330447535, 2024-06-14:
Add the following code to solve the problem of not opening the page under SSH，View the api [https://code.visualstudio.com/api/advanced-topics/remote-extensions#forwarding-localhost](asExternal)

`
    const pageUrl = await vscode.env.asExternalUri(
      vscode.Uri.parse(pioNodeHelpers.home.getFrontendUrl({
        start: startUrl,
        theme,
        workspace: extension.getEnterpriseSetting('defaultPIOHomeWorkspace'),
      }))
    );
`

Adib6637, 2025-02-01:
hello everybody,  I solve the problem by enabling X11 forwarding. my .ssh/config look like this:

Host xxxxx
  HostName xxxxxx
  User xxxxxx
  LocalForward 127.0.0.1:8008 127.0.0.1:8008
  Port 22
  PreferredAuthentications publickey 
  IdentityFile xxxxxxx
  ForwardX11 yes
  ForwardX11Trusted yes