platformio/platformio-vscode-ide #946: Cpp code formater

porlock, 2019-08-02:
Hey, i have a problem with cpp code formater, i don’t know why but i have more than one. I don’t install anything more than C/C++ for Visual Studio Code. When i run VSC and open cpp file in platformio alt+shift+f doesn’t work, have to use ctl+shift+p and select “format document with…” and have multipe choice (please see attachemnt) and only one works, random one, after some time i get more and more formaters to choice, all named the same and like previously only one works. Please help.
[image]

pfeerick, 2019-08-06:
Not PlatformIO related... it's a vscode-cpptools bug that has had open issues since March and still no resolution. 

microsoft/vscode-cpptools#3285

ivankravets, 2019-08-11:
Duplicate of microsoft/vscode-cpptools#3285

closed 2019-08-11 by ivankravets