platformio/platformio-vscode-ide #950: Registry Search Libraries

Galaxy-Man, 2019-08-05:
Issue Type: <b>Bug</b>

I am using VSC Platformio Arduino.

When selecting the 'Search libraries' box cursor does not do anything you cannot select anything and you cannot type into the box. 

All other clickable items work.

Any ideas?

Extension version: 1.8.1
VS Code version: Code 1.36.1 (2213894ea0415ee8c85c5eea0d0ff81ecc191529, 2019-07-08T22:59:35.033Z)
OS version: Windows_NT x64 10.0.17134

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz (8 x 2592)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_deferred_display_list: disabled_off<br>skia_renderer: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.91GB (9.03GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

pfeerick, 2019-08-06:
Sounds like a duplicate of https://github.com/platformio/platformio-vscode-ide/issues/892#issuecomment-514553283

Check out that thread to see the underlying issue, and the suggested workaround until the VSCode team fix it.

ivankravets, 2019-08-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/892

closed 2019-08-11 by ivankravets