platformio/platformio-vscode-ide #954: Installation Manager

dusan06, 2019-08-08:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.36.1
PIO IDE: v1.8.1
System: Linux, 5.0.0-23-generic, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/hassio/.platformio/.cache/tmp-19422FzGaASVER8NW/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/hassio/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:31159)
```

JBFUK, 2019-08-09:
+1

%23 Configuration

VSCode: 1.37.0
PIO IDE: v1.8.1
System: Linux, 4.15.0-55-generic, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/justin/.platformio/.cache/tmp-28389Jf0c6twiB3VX/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/justin/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:31159)

ivankravets, 2019-08-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/907

closed 2019-08-11 by ivankravets