platformio/platformio-vscode-ide #957: The latest VSCode 1.37 seems to break PlatformIO

ssilverman, 2019-08-09:
I'm still diving into this, but when I just upgraded to VSCode v1.37.0, it broke something about platformio paths:

```
> Executing task: platformio run <

The terminal shell path "platformio" does not exist
```

ssilverman, 2019-08-09:
I tried a reinstall of the PlatformIO extension, but that doesn't work either.

ssilverman, 2019-08-09:
I just tried deleting my ~/.platformio directory, and after it rebuilt itself, I still get the same error.

valeros, 2019-08-09:
Hi @ssilverman! What is your OS? Can't reproduce the issue on Windows machine.

ssilverman, 2019-08-09:
I also just tried setting this in my custom settings, and this didn't work either:
```
"platformio-ide.customPATH": "/Users/shawn/.platformio/penv/bin:/Users/shawn/.platformio/penv"
```

My OS is OSX 10.14.6.

ssilverman, 2019-08-09:
I verified I don't have other platformio's in my path, and also verified that there are none installed via `brew` or `pip`.

ssilverman, 2019-08-09:
Ok, I just completely deleted the app plus ~/.platformio then did a reinstall. No dice :(

ssilverman, 2019-08-09:
New detail: I can open a VSCode terminal and run commands manually from there, that works fine. It's when running anything PlatformIO from the "Run Task..." list.

valeros, 2019-08-09:
Thanks for the info, it seems like v1.37 breaks the extension on Unix based OS. As a workaround, you can globally install PlatformIO until we fix it.

pczekalski, 2019-08-09:
Same here. Ubuntu 19.04, VS 1.3.7, PIO 4.0.0.
Could you give some details on what "you can globally install PlatformIO" means? Installing via pip nor pip3 does not help, either.

valeros, 2019-08-09:
Forget to mention that you also need to disable the `Use Builtin PIOCore` option in 
`Settings->Extensions->PlatformIO IDE Configuration`

pczekalski, 2019-08-09:
Thx - it works this way (at least compiles).

SigmundurJohansen, 2019-08-09:
Is there a fix comming?

sstaub, 2019-08-09:
I have the same problem with macOS Catalina beta, on Mojave VSCode 1.37 works without a problem.

tuxBurner, 2019-08-09:
Have the same issue https://github.com/MarlinFirmware/Marlin/issues/14887#issuecomment-519994880

Tatukis, 2019-08-09:
> Same here. Ubuntu 19.04, VS 1.3.7, PIO 4.0.0.
> Could you give some details on what "you can globally install PlatformIO" means? Installing via pip nor pip3 does not help, either.

I just use the SuperQuick install in the plataformio page (python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)")

ryanburketllc, 2019-08-09:
Uninstalling and reinstalling PlatformIO from the vscode extensions menu didn't do anything for me, but following Tatukis' advice, the SuperQuick installer for mac running from within the vscode terminal resolved it for me. Made for a frustrating morning though... https://docs.platformio.org/en/latest/installation.html#super-quick-mac-linux

joaodematejr, 2019-08-09:
+1 Mac Os Mojave 10.14.6

giotherobot, 2019-08-09:
Confirmed also on Linux Debian Buster.
Confirmed also that the problem is in the Tasks, using platformio thru cli still works with the integrated version, no need to install system-wide.

pfeerick, 2019-08-10:
Perils of updating as soon as it's released... Should know better with Micro$haft... always wait for the second revision! 😆 Can confirm it isn't causing issues on my Windows 10, not sure about Linux yet, not updating to 1.37 on my daily until this is fixed!

zerog2k, 2019-08-10:
on mac, after rolling back to 1.36.1
(https://code.visualstudio.com/updates/v1_36)

it's working again, but running platformio build, I see this ominous message:
>Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead

```
> Executing task in folder mysensors-promini-test1: platformio run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Warning! Ignore unknown configuration option `lib_deps` in section [platformio]
DEPRECATED! A legacy library storage `/Users/jens/Documents/PlatformIO/Projects/mysensors-promini-test1/.piolibdeps` has been found in a project. 
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `/Users/jens/Documents/PlatformIO/Projects/mysensors-promini-test1/.piolibdeps` folder.
More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
```

phhoef, 2019-08-10:
Same here for VSCode.
It seems, that the build/upload task won't open the PlatformIO Shell properly.
When clicking in the bottom bar to open a shell window and manually executing `platformio run --target upload` it is working fine. So I guess the installation is not completely broken ...

kbhuinfo, 2019-08-11:
> > Same here. Ubuntu 19.04, VS 1.3.7, PIO 4.0.0.
> > Could you give some details on what "you can globally install PlatformIO" means? Installing via pip nor pip3 does not help, either.
> 
> I just use the SuperQuick install in the plataformio page (python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)")
> EDIT: Probably with sudo.

That is the only method, that worked for me (kubuntu 19.04 / vsc 1.37 / Pio4.0.0)

ivankravets, 2019-08-11:
This is a bug of VSCode. It's 2nd time when a new release of VSCode breaks Tasks API. The current workaround is to downgrade VSCode to 1.36.

@alexr00, @Tyriar, @dbaeumer could you explain how to fix it now? We use `ProcessExecution` API, see https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L68

ivankravets, 2019-08-11:
Please update to https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.2

We switched temporarily to VSCode ShellExecution API.

taburet, 2019-08-11:
Thanks for the super quick fix!

Tyriar, 2019-08-11:
@alexr00 might be a candidate for recovery build for `ProcessExecution`?

SigmundurJohansen, 2019-08-11:
@ivankravets  updating to 1.8.2 did not fix it for me

ivankravets, 2019-08-11:
@SigmundurJohansen what is your OS and default shell?

SigmundurJohansen, 2019-08-11:
Ubuntu 18.04.2 LTS
bin/bash

ivankravets, 2019-08-11:
I don’t have more ideas for workarounds. Let’s wait for the official tips from VSCode team. 

We could actually try to play with system shells and detect it automatically. However, this is “dirty way”.

sstaub, 2019-08-11:
Now it works with macOS Catalina beta.

giotherobot, 2019-08-11:
Works fine on Debian Buster.
Thanks for quick fix!

pfeerick, 2019-08-11:
v1.82 works fine on a KUbuntu VM and my daily Manjaro Arch with VSCode 1.37.

alexr00, 2019-08-12:
@ivankravets and @Tyriar thank you for tagging me. Yes, this is a candidate for the recovery build. Corresponding VS Code issue: https://github.com/microsoft/vscode/issues/78898

ivankravets, 2019-08-12:
@alexr00 thanks for the quick fix! What is the right API in our case? Should we use `ProcessExecution` or `ShellExecution ` if we call `platformio.exe` binary? It seems like it should be `ProcessExecution`.

alexr00, 2019-08-12:
Since you are calling `platformio.exe` `ProcessExecution` is a better fit.

alexr00, 2019-08-12:
The fix will go out in 1.37.1 and will also be available in the next insiders build.

ivankravets, 2019-08-12:
@alexr00 Thanks! We will revert back to `ProcessExecution` when `1.37.1` is out.

ivankravets, 2019-08-12:
Do you have any ETA for 1.37.1?

Tyriar, 2019-08-12:
@ivankravets likely mid-late this week

ivankravets, 2019-08-16:
@SigmundurJohansen does it work with the latest VSCode 1.37.1? It's already released.

closed 2019-08-16 by ivankravets

pczekalski, 2019-08-17:
No, it does not. Fails in installing core.

pfeerick, 2019-08-17:
@pczekalski That sounds to be completely different issue as this one is only related to ability to run commands via the PlatformIO extension, not installation (or perhaps reinstallation in your case?) of PlatformIO via the extension. 

On VSCode 1.37.0, with v1.81 of the extension (pre-workaround) I can confirm it was broken on KUbuntu. I then updated to VSCode 1.37.1 and the issue is now gone. Sweet!

ivankravets, 2019-08-17:
@pczekalski please file a new issue here https://github.com/platformio/platformio-vscode-ide/issues with all details.

commit cb5b8b60 (2019-08-11): Temporary workaround for the broken Tasks API in the latest VSCode 1.37 // Resolve #957
commit 6e92f307 (2019-08-16): Revert back to `vscode.ProcessExecution` for Tasks // Resolve #957