platformio/platformio-vscode-ide #959: Start PIO Home Server

GarrettButchko, 2019-08-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.0
PIO IDE: v1.8.2
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/garrettbutchko/.vscode/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:17225)
	at ChildProcess.h (/Users/garrettbutchko/.vscode/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:6033)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-08-31:
What is your Python version? Did you install it manually? See related post 
 https://community.platformio.org/t/cant-open-pio-home/8588/22?u=ivankravets

Theremingenieur, 2019-09-19:
Apparently, Apple did some drop in Python support in macOS Catalina. Thus, everything depends from previously installed python versions (if any). Installing python 3.7.4 (the newest one) from python.org and deleting the penv folder did the trick for me.

ivankravets, 2019-10-11:
Please upgrade PlatformIO IDE for VSCode to 1.9.0

closed 2019-10-11 by ivankravets