platformio/platformio-vscode-ide #967: Installation Manager

JoemioF, 2019-08-13:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.0
PIO IDE: v1.8.2
System: Linux, 5.0.0-23-generic, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/joemio-jr/.platformio/.cache/tmp-4185lMngXxIZRDMU/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/joemio-jr/.vscode/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:31185)
```

ivankravets, 2019-08-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/962

ivankravets, 2019-08-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/907

closed 2019-08-28 by ivankravets