platformio/platformio-vscode-ide #970: Tasks don't work if integrated shell is set to wsl

Tiim, 2019-08-14:
I usually use the [WSL](https://de.wikipedia.org/wiki/Windows_Subsystem_for_Linux) shell set in VSCode as my default shell. 

```js
// settings.json
{
"terminal.integrated.shell.windows": "C:\\WINDOWS\\System32\\wsl.exe",
}
```

When I'm trying to run any tasks i get the following output;

```
> Executing task: platformio.exe run <

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

When I change the shell to `cmd.exe` the tasks work,

ivankravets, 2019-08-16:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/957

ivankravets, 2019-08-16:
Please update VScode to 1.37.1.

Does it work now?

closed 2019-08-16 by ivankravets

Tiim, 2019-08-20:
I do have version 1.37 installed and still get the error:

[image]