platformio/platformio-vscode-ide #979: Selecting board options

GianniDPC, 2019-08-20:
Why is there no easy way of selecting board options like you have with the Arduino plugin for vscode?

For example for my BluePill I have to select HID bootloader and some USB options to make the device appear as a COM-port.

Now to be able to to this I have to manually add some parameters to the build options, whereas with the Arduino plugin I could just select the right configuration from "board configuration".

ivankravets, 2019-08-31:
How does look your `platformio.ini` after configuration?

lnlp, 2019-09-06:
This is actually a request for a (G)UI where all board parameters and possible values can be easily selected from a menu (for the selected board). 
I agree that this selection process works easier in the Arduino IDE than it currently does in PlatformIO.

ivankravets, 2021-01-04:
You need to declare different project environments and use switcher for them https://docs.platformio.org/en/latest/integration/ide/vscode.html#task-explorer

closed 2021-01-04 by ivankravets

lnlp, 2021-03-22:
> You need to declare different project environments and use switcher for them https://docs.platformio.org/en/latest/integration/ide/vscode.html#task-explorer

I'm not sure why you posted that as answer. It is not a solution for the original question.