platformio/platformio-vscode-ide #984: Visual Studio Code with platform IO screen updates are problematic

bertdd, 2019-08-22:
I became curious to use PlatformIO with Visual Studio Code for ESP32 development as an alternative for "big" Visual Studio with Visual Micro.
So far I have been disappointed with the combi VSCode and PlatformIO because on my system, a 15 Inch Macbook Pro with Windows 10 fast lane in dual boot  the display of VSCode does not work properly. Areas on my screen are simply not updated properly which makes basic operation allmost impossible: 

[image]

This is a typical example, no idea where it comes from, no other program on my system shows this behaveour. It seems to be better with later releases, but the issue is not gone completely.

ivankravets, 2019-08-25:
Does it work in a browser if you click on PIO Home `resize` icon on the top toolbar (near Twitter icon). It should open PIO Home standalone browser.

ivankravets, 2019-08-31:
Do you have any updates?

bertdd, 2019-08-31:
yes, when running it in the browser there are no display issues, but then i cannot find a project explorer...

(maybe this is my Visual Studio expectation?)

ivankravets, 2019-08-31:
Do you use the latest version of VSCode?

closed 2019-11-16 by ivankravets