platformio/platformio-vscode-ide #994: Installation Manager

zz89419318, 2019-08-28:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.1
PIO IDE: v1.8.2
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\89411\.platformio\.cache\tmp-17984JIkjL2sZUFo2\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\89411\.platformio\.cache\tmp-17984JIkjL2sZUFo2\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\89411\.platformio\.cache\tmp-17984JIkjL2sZUFo2\virtualenv-16.1.0\src\virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\89411\.platformio\.cache\tmp-17984JIkjL2sZUFo2\virtualenv-16.1.0\src\virtualenv.py", line 1317, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Python27\lib\shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\89411\\.platformio\\penv\\Scripts\\python.exe'

New python executable in C:\Users\89411\.platformio\penv\Scripts\python.exe

	at t.value (C:\Users\89411\.vscode\extensions\platformio.platformio-ide-1.8.2\node_modules\platformio-node-helpers\dist\index.js:1:31185)
```

jedmao, 2019-08-31:
Same here, using WSL on Windows 10:

## Configuration

VSCode: 1.37.1
PIO IDE: v1.8.2
System: Linux, 4.4.0-17763-Microsoft, x64

## Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/jedmao/.platformio/.cache/tmp-190KnuEnwm77x7y/virtualenv-16.1.0/src/virtualenv.py", line 20, in <module>
    import distutils.spawn
ModuleNotFoundError: No module named 'distutils.spawn'

	at t.value (/home/jedmao/.vscode-server/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:31092)
	at process._tickCallback (internal/process/next_tick.js:68:7)
```

ivankravets, 2019-08-31:
@jedmao  https://github.com/platformio/platformio-vscode-ide/issues/907

ivankravets, 2019-08-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/763

closed 2019-08-31 by ivankravets