platformio/platformio-vscode-ide #995: Start PIO Home Server

ruben-vc, 2019-08-29:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.1
PIO IDE: v1.8.2
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/rubenvc/.vscode/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:17225)
	at ChildProcess.h (/Users/rubenvc/.vscode/extensions/platformio.platformio-ide-1.8.2/node_modules/platformio-node-helpers/dist/index.js:1:6033)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ruben-vc, 2019-08-29:
Problem installing platformio in vscode (mac catalina 10.15)

ivankravets, 2019-08-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

ivankravets, 2019-08-31:
Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

closed 2019-08-31 by ivankravets

ruben-vc, 2019-09-01:
Install pip with:  sudo easy_install pip
Install the last python version from python.org  (3.7.4)
Remove directory (it's a hidden directory) ~/.platformio/penv 
Everything working!!! Thanks!!