platformio/platformio-vscode-ide #997: Device connected over Micro USB disconnects when debugger starts

mirao, 2019-08-31:
**Steps to reproduce:**

- Have connected MCU STM32F407-DISC1 to PC host according to https://github.com/mirao/stm32f4discovery/blob/master/stm32cube-virtualCOMPort/README.md:
```
[10665.483428] usb 1-1: new full-speed USB device number 42 using xhci_hcd
[10665.633703] usb 1-1: New USB device found, idVendor=0483, idProduct=374b, bcdDevice= 1.00
[10665.633711] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10665.633716] usb 1-1: Product: STM32 STLink
[10665.633722] usb 1-1: Manufacturer: STMicroelectronics
[10665.633726] usb 1-1: SerialNumber: 066CFF565257867767193452
[10665.691081] usb-storage 1-1:1.1: USB Mass Storage device detected
[10665.692888] scsi host3: usb-storage 1-1:1.1
[10665.693769] cdc_acm 1-1:1.2: ttyACM0: USB ACM device
[10666.724877] scsi 3:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
[10666.726198] sd 3:0:0:0: Attached scsi generic sg1 type 0
[10666.727901] sd 3:0:0:0: [sdb] 2120 512-byte logical blocks: (1.09 MB/1.04 MiB)
[10666.728189] sd 3:0:0:0: [sdb] Write Protect is off
[10666.728196] sd 3:0:0:0: [sdb] Mode Sense: 03 00 00 00
[10666.728411] sd 3:0:0:0: [sdb] No Caching mode page found
[10666.728422] sd 3:0:0:0: [sdb] Assuming drive cache: write through
[10666.748321] sd 3:0:0:0: [sdb] Attached SCSI removable disk
[10668.219710] usb 1-2: new full-speed USB device number 43 using xhci_hcd
[10668.369926] usb 1-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[10668.369935] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10668.369940] usb 1-2: Product: STM32 Virtual ComPort
[10668.369944] usb 1-2: Manufacturer: STMicroelectronics
[10668.369948] usb 1-2: SerialNumber: 3274376C3437
[10668.372752] cdc_acm 1-2:1.0: ttyACM1: USB ACM device
```
```
$ ls -ld /dev/ttyACM*
crw-rw-rw- 1 root plugdev 166, 0 Aug 31 20:21 /dev/ttyACM0
crw-rw-rw- 1 root dialout 166, 1 Aug 31 20:22 /dev/ttyACM1
```
- Build and upload the app from the mentioned project to MCU
- Transfer over virtual COM port works well. Tested with PlatformIO's serial monitor and also directly using `echo 'o' > /dev/ttyACM1` and `tail -f /dev/ttyACM1`
- Run debugger (F5) in VSCode

**Expected result:**
- USB device "STM32 Virtual ComPort" is still connected

**Actual result:**
- USB device "STM32 Virtual ComPort" is disconnected. Only STLink stays connected
```
$ dmesg
[11383.546701] usb 1-2: USB disconnect, device number 43
[11383.546779] cdc_acm 1-2:1.0: failed to set dtr/rts
```
```
$ ls -ld /dev/ttyACM*
crw-rw-rw- 1 root plugdev 166, 0 Aug 31 20:21 /dev/ttyACM0
```
platformio.ini:
```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = stm32cube
monitor_port = /dev/ttyACM1

[platformio]
src_dir = Src
lib_dir = Middlewares/ST
```
debug log: https://gist.github.com/mirao/30c026ee119de72a6cd8c12617d7d7f3

**Used SW/HW:**

- VSCode 1.37.1
- PIO 4.0.2
- PIO IDE 1.8.3
- MCU STM32F407-DISC1

ivankravets, 2019-10-11:
Have you resolved this issue?

mirao, 2019-10-12:
It's strange, now it's behaving unpredictably. It's true that the virtual device is always disconnected in debug mode, but then:
Sometimes the device is connected again when I press F5 (Continue).
```
[11647.459608] usb 1-1: USB disconnect, device number 14
[11662.060121] usb 1-1: new full-speed USB device number 15 using xhci_hcd
[11662.210612] usb 1-1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[11662.210619] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11662.210624] usb 1-1: Product: STM32 Virtual ComPort
[11662.210627] usb 1-1: Manufacturer: STMicroelectronics
[11662.210631] usb 1-1: SerialNumber: 3274376C3437
[11662.212907] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
``` 
but occasionally the device keeps disconnected and OS shows errors about it:
```
[11238.586933] usb 1-1: USB disconnect, device number 9
[11238.587027] cdc_acm 1-1:1.0: failed to set dtr/rts
[11257.473443] usb 1-1: new full-speed USB device number 10 using xhci_hcd
[11273.017514] usb 1-1: device descriptor read/64, error -110
[11288.633330] usb 1-1: device descriptor read/64, error -110
[11288.869261] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[11298.489134] usb 1-1: device descriptor read/all, error -110
[11298.489256] usb usb1-port1: attempt power cycle
[11299.141166] usb 1-1: new full-speed USB device number 12 using xhci_hcd
[11304.377235] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[11304.585339] usb 1-1: Device not responding to setup address.
[11304.793134] usb 1-1: device not accepting address 12, error -71
[11304.921023] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[11310.008938] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[11310.217117] usb 1-1: Device not responding to setup address.
[11310.424857] usb 1-1: device not accepting address 13, error -71
[11310.424989] usb usb1-port1: unable to enumerate USB device
```
When I stop debug mode, the device is connected again (always).

mirao, 2019-10-12:
Well, it's actually correct that the virtual device is connected only when it is initialized in app (in function `MX_USB_DEVICE_Init`). It means that debugger must reach a call of that function to get visible the device in host OS.
So only issue is why the device occasionally fails to connect. I must admit that now it's working well most of time.

mirao, 2019-10-12:
Closing, now it's working. Anybody feel free to reopen if you hit the issue.

closed 2019-10-12 by mirao