platformio/platformio-vscode-ide #998: Cannot enter search terms

mderouss, 2019-08-31:
Issue Type: <b>Bug</b>

Using Xubuntu 19.04. With PIO installed but no project defined, navigate to  screen for installing an embedded platform. Attempts to enter text in the box containing 'Filter platforms by name...' fail. Similar behaviour when  attempting to search for a board - no search term can be entered. 

Extension version: 1.8.3
VS Code version: Code 1.37.1 (f06011ac164ae4dc8e753a3fe7f9549844d15e35, 2019-08-15T16:17:25.463Z)
OS version: Linux x64 5.0.0-25-generic

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Xeon(R) CPU           X5675  @ 3.07GHz (24 x 1829)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: disabled_software<br>skia_deferred_display_list: disabled_off<br>skia_renderer: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: unavailable_off<br>viz_display_compositor: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|0, 0, 0|
|Memory (System)|47.14GB (44.04GB free)|
|Process Argv|--unity-launch|
|Screen Reader|no|
|VM|0%|
</details>

The-Experimentalist, 2019-09-07:
Same issue here with various text entry boxes, today I cannot enter search terms in any of the library manager search text boxes

Extension version: 1.8.3

Visual Studio Code:
Version: 1.38.0 (user setup)
Commit: 3db7e09f3b61f915d03bbfa58e258d6eee843f35
Date: 2019-09-03T21:49:13.459Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.17763

Restarting VSCode fixes  it but it is very annoying

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/892

closed 2019-10-11 by ivankravets