platformio/platformio-vscode-ide #1002: I installed Anaconda on my computer, now PlatformIO won't update or install

fivefishstudios, 2019-09-04:
I installed Anaconda on my computer, now PlatformIO won't update or install.
I tried uninstalling PlatformIO IDE VScode extension and re-installing it and same problem. 

Error: Conda Virtualenv: 
[Errno 13] Permission denied: '/.cph_tmp64gtq87s'
[Errno 13] Permission denied: '/.cph_tmpex5bab12'

    at f.value.o.runCommand (/Users/rueloquindo/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:29301)
    at ChildProcess.h (/Users/rueloquindo/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/misc.js:172:5)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
    at Socket.emit (events.js:182:13)
    at Pipe.Socket._destroy._handle.close (net.js:606:12)

fivefishstudios, 2019-09-04:
FIX: I uninstalled PlatformIO, uninstalled Anaconda3.
Then re-installed PlatformIO, and re-installed Anaconda3. 

Everything seems to be back to normal now.

closed 2019-09-04 by fivefishstudios

QGB, 2022-04-23:
```Please wait...
The first initialization requires Internet connection and may take a while (need to install dependent toolchains, frameworks, SDKs).
Please be patient and let the initialization complete.```