platformio/platformio-vscode-ide #1007: Installation Manager

JohannesDosReis, 2019-09-06:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.0
PIO IDE: v1.8.3
System: Linux, 5.2.11-1-MANJARO, x64

%23 Exception
```
Error: PIP Core: ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.

	at f.value.o.runCommand (/home/johannes/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:32222)
	at ChildProcess.h (/home/johannes/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

DirkHeinke, 2019-09-08:
Did you find a solution? I have the exact same problem.

JohannesDosReis, 2019-09-09:
> Did you find a solution? I have the exact same problem.

Yes, just updatethe pip

exetico, 2019-09-15:
@JohannesDosReis  - `pip install --upgrade pip` or? 

I simply can't get this to work :( ...

ivankravets, 2019-10-11:
What is your?
```
python --version
python3 --version
virtualenv --version
pip --version
pip3 --version
```

ivankravets, 2019-10-11:
Hi all,

Could you try the upcoming PlatformIO IDE for VSCode 1.9.0. See instruction in "Solution" section here https://github.com/platformio/platformio-vscode-ide/issues/959

ivankravets, 2019-10-12:
@pfeerick could you help me to resolve this issue? I'll need more info here.

Yes, this an issue of MANJARO, they force PIP to `user` folder even if we call PIP from a virtual environment. It's total breaks Python development on MANJARO. I don't know how Python devs use this Linux version.

I think we can find a solution here.

1. Could you revert back `user = true`.
2. As I understand, `python3` is already pre-installed? Try the next:
```
python3 -m venv /tmp/piotest
source /tmp/piotest/bin/activate
pip list
```

pfeerick, 2019-10-12:
> I don't know how Python devs use this Linux version.

Good question... although I think the same can be said about the actual Python devs at times :-P 

I just mentioned something about this on the Manjaro forum, and it seems like it's going to get dropped, as there has been no traction to fix the underlying issues in pip itself. 

> Just downgrade python-pip to whatever is now in stable. I've now abandoned trying to help upstream.

```
[pfeerick@asus-h55d etc]$ python3 --version
Python 3.7.4
[pfeerick@asus-h55d etc]$ cat /etc/pip.conf 
[global]
user = true
[pfeerick@asus-h55d etc]$ rm -R /tmp/piotest/
[pfeerick@asus-h55d etc]$ python3 -m venv /tmp/piotest
[pfeerick@asus-h55d etc]$ source /tmp/piotest/bin/activate
(piotest) [pfeerick@asus-h55d etc]$ pip list
(piotest) [pfeerick@asus-h55d etc]$ ls -la /tmp/piotest
total 4
drwxr-xr-x  5 pfeerick pfeerick 140 Oct 12 20:20 .
drwxrwxrwt 24 root     root     620 Oct 12 20:20 ..
drwxr-xr-x  2 pfeerick pfeerick 240 Oct 12 20:20 bin
drwxr-xr-x  2 pfeerick pfeerick  40 Oct 12 20:20 include
drwxr-xr-x  3 pfeerick pfeerick  60 Oct 12 20:20 lib
lrwxrwxrwx  1 pfeerick pfeerick   3 Oct 12 20:20 lib64 -> lib
-rw-r--r--  1 pfeerick pfeerick  69 Oct 12 20:20 pyvenv.cfg
```

ivankravets, 2019-10-12:
This is interesting :) They remove `pip` from official Python 3.7 installation or didn't include it in a default distribution. Can you install PIP in this virtual env?
```
(piotest) python get-pip.py
```

See https://pip.pypa.io/en/stable/installing/

So, try now `pip install platformio` in this virt. env.

pfeerick, 2019-10-12:
Looks like we're back to the same problem. 

```
(piotest) [pfeerick@asus-h55d piotest]$ curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1733k  100 1733k    0     0  1033k      0  0:00:01  0:00:01 --:--:-- 1033k
(piotest) [pfeerick@asus-h55d piotest]$ python get-pip.py
Collecting pip
  Using cached https://files.pythonhosted.org/packages/30/db/9e38760b32e3e7f40cce46dd5fb107b8c73840df38f0046d8e6514e675a1/pip-19.2.3-py2.py3-none-any.whl
Collecting wheel
  Using cached https://files.pythonhosted.org/packages/00/83/b4a77d044e78ad1a45610eb88f745be2fd2c6d658f9798a15e384b7d57c9/wheel-0.33.6-py2.py3-none-any.whl
Installing collected packages: pip, wheel
Successfully installed pip-19.2.3 wheel-0.33.6
(piotest) [pfeerick@asus-h55d piotest]$ pip install platformio
Ignoring '--user', since user site-packages are not visible in this virtualenv.
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/5d/59/2ba54e3f2ae8435997acbd0d0ff31993a7041a8b11515e0b5cc846a71ca4/platformio-4.0.3.tar.gz
Collecting bottle<0.13 (from platformio)
  Using cached https://files.pythonhosted.org/packages/69/d1/efdd0a5584169cdf791d726264089ce5d96846a8978c44ac6e13ae234327/bottle-0.12.17-py3-none-any.whl
Collecting click<8,>=5 (from platformio)
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Collecting colorama (from platformio)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting pyserial!=3.3,<4,>=3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio)
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting semantic_version<3,>=2.8.1 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0f/3b/8fee26649a86c71df159ed0ae7ac5f9ac38829bccd8a7404e116f903929b/semantic_version-2.8.2-py2.py3-none-any.whl
Collecting tabulate<1,>=0.8.3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/66/d4/977fdd5186b7cdbb7c43a7aac7c5e4e0337a84cb802e154616f3cfc84563/tabulate-0.8.5.tar.gz
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/18/b0/8146a4f8dd402f60744fa380bc73ca47303cccf8b9190fd16a827281eac2/certifi-2019.9.11-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/e0/da/55f51ea951e1b7c63a579c09dd7db825bb730ec1fe9c0180fc77bfb31448/urllib3-1.25.6-py2.py3-none-any.whl
Collecting idna<2.9,>=2.5 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Installing collected packages: bottle, click, colorama, pyserial, certifi, chardet, urllib3, idna, requests, semantic-version, tabulate, platformio
  Running setup.py install for tabulate ... done
  Running setup.py install for platformio ... done
Successfully installed bottle-0.12.17 certifi-2019.9.11 chardet-3.0.4 click-7.0 colorama-0.4.1 idna-2.8 platformio pyserial-3.4 requests-2.22.0 semantic-version-2.8.2 tabulate urllib3-1.25.6
(piotest) [pfeerick@asus-h55d piotest]$ platformio
Traceback (most recent call last):
  File "/home/pfeerick/.local/bin/platformio", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/tmp/piotest/lib/python3.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==4.0.3' distribution was not found and is required by the application
(piotest) [pfeerick@asus-h55d piotest]$ date
Sat Oct 12 20:33:30 AEST 2019
(piotest) [pfeerick@asus-h55d piotest]$ ls -la /home/pfeerick/.local/
total 20
drwxr-xr-x  5 pfeerick pfeerick 4096 Oct 12 20:31 .
drwxr-xr-x 54 pfeerick pfeerick 4096 Oct 12 20:19 ..
drwxr-xr-x  2 pfeerick pfeerick 4096 Oct 12 20:32 bin
drwxr-xr-x  3 pfeerick pfeerick 4096 Oct 12 20:31 lib
[... cut ...]
  debug     PIO Unified Debugger
  device    Monitor device or list existing
  home      PIO Home
  init      Initialize PlatformIO project or update existing
  lib       Library Manager
  platform  Platform Manager
  remote    PIO Remote
  run       Process project environments
  settings  Manage PlatformIO settings
  test      Unit Testing
  update    Update installed platforms, packages and libraries
  upgrade   Upgrade PlatformIO to the latest version
(piotest) [pfeerick@asus-h55d piotest]$ 
```

ivankravets, 2019-10-12:
Could you repeat the same but without installing PIP? I mean using system provided but with a custom `pip.conf`.

pfeerick, 2019-10-12:
Oh, and pip is present in the venv... a directory listing of the `bin/` folder shows it and python present. On a new venv:

```
(piotest2) [pfeerick@asus-h55d piotest2]$ ls bin
activate      activate.fish  easy_install-3.7  pip3    python
activate.csh  easy_install   pip               pip3.7  python3
```

With provided pip, but no custon pip.conf

```
(piotest2) [pfeerick@asus-h55d piotest2]$ pip install platformio
Ignoring '--user', since user site-packages are not visible in this virtualenv.
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/5d/59/2ba54e3f2ae8435997acbd0d0ff31993a7041a8b11515e0b5cc846a71ca4/platformio-4.0.3.tar.gz
Collecting bottle<0.13 (from platformio)
  Using cached https://files.pythonhosted.org/packages/69/d1/efdd0a5584169cdf791d726264089ce5d96846a8978c44ac6e13ae234327/bottle-0.12.17-py3-none-any.whl
Collecting click<8,>=5 (from platformio)
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Collecting colorama (from platformio)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting pyserial!=3.3,<4,>=3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio)
  Using cached https://files.pythonhosted.org/packages/51/bd/23c926cd341ea6b7dd0b2a00aba99ae0f828be89d72b2190f27c11d4b7fb/requests-2.22.0-py2.py3-none-any.whl
Collecting semantic_version<3,>=2.8.1 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0f/3b/8fee26649a86c71df159ed0ae7ac5f9ac38829bccd8a7404e116f903929b/semantic_version-2.8.2-py2.py3-none-any.whl
Collecting tabulate<1,>=0.8.3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/66/d4/977fdd5186b7cdbb7c43a7aac7c5e4e0337a84cb802e154616f3cfc84563/tabulate-0.8.5.tar.gz
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/e0/da/55f51ea951e1b7c63a579c09dd7db825bb730ec1fe9c0180fc77bfb31448/urllib3-1.25.6-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/18/b0/8146a4f8dd402f60744fa380bc73ca47303cccf8b9190fd16a827281eac2/certifi-2019.9.11-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting idna<2.9,>=2.5 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/14/2c/cd551d81dbe15200be1cf41cd03869a46fe7226e7450af7a6545bfc474c9/idna-2.8-py2.py3-none-any.whl
Installing collected packages: bottle, click, colorama, pyserial, urllib3, certifi, chardet, idna, requests, semantic-version, tabulate, platformio
  Running setup.py install for tabulate ... done
  Running setup.py install for platformio ... done
Successfully installed bottle-0.12.17 certifi-2019.9.11 chardet-3.0.4 click-7.0 colorama-0.4.1 idna-2.8 platformio pyserial-3.4 requests-2.22.0 semantic-version-2.8.2 tabulate urllib3-1.25.6
(piotest2) [pfeerick@asus-h55d piotest2]$ platformio
Traceback (most recent call last):
  File "/home/pfeerick/.local/bin/platformio", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/tmp/piotest2/lib/python3.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==4.0.3' distribution was not found and is required by the application
(piotest2) [pfeerick@asus-h55d piotest2]$ ls -la ~/.local
total 20
drwxr-xr-x  5 pfeerick pfeerick 4096 Oct 12 20:48 .
drwxr-xr-x 54 pfeerick pfeerick 4096 Oct 12 20:19 ..
drwxr-xr-x  2 pfeerick pfeerick 4096 Oct 12 20:48 bin
drwxr-xr-x  3 pfeerick pfeerick 4096 Oct 12 20:48 lib
drwx------ 26 pfeerick pfeerick 4096 Oct 12 19:44 share
(piotest2) [pfeerick@asus-h55d piotest2]$ date
Sat Oct 12 20:48:49 AEST 2019
```

With system provided pip, WITH pip.conf user=no
[... 41 lines cut ...]

Options:
  --version          Show the version and exit.
  -f, --force        Force to accept any confirmation prompts.
  -c, --caller TEXT  Caller ID (service).
  -h, --help         Show this message and exit.

Commands:
  account   Manage PIO Account
  boards    Embedded Board Explorer
  ci        Continuous Integration
  debug     PIO Unified Debugger
  device    Monitor device or list existing
  home      PIO Home
  init      Initialize PlatformIO project or update existing
  lib       Library Manager
  platform  Platform Manager
  remote    PIO Remote
  run       Process project environments
  settings  Manage PlatformIO settings
  test      Unit Testing
  update    Update installed platforms, packages and libraries
  upgrade   Upgrade PlatformIO to the latest version
(piotest3) [pfeerick@asus-h55d piotest3]$ 
```

ivankravets, 2019-10-12:
A workaround is added in https://github.com/platformio/platformio-node-helpers/commit/4760823f20150036bf25731601055ffa4713088f

Please re-test with 1.9.1-beta 

1. Download https://www.dropbox.com/s/d37kbuai6eryugh/platformio-ide-1.9.1-beta.1.vsix?dl=1
2. VSCode > View > Command Palette > Install from VSIX.

Try to revert back default Manjaro settings. Does it work now?

pfeerick, 2019-10-12:
I ensured that the pip.conf was back to the default:
```
[pfeerick@asus-h55d etc]$ cat /etc/pip.conf
[global]
user = true
```

Installed the platformio-ide-1.9.1-beta.1.vsix 

It installed PlatformIO correctly, and PlatformIO started up properly on the restart of VSCode. 

Thanks for the prompt fix. 

It's annoying that you had to do this... especially as I've just found out that if we enable the 'downgrade packages' option in the pamac-manager (or run `pamac -Syyuu`) it'll offer to revert to a working version of python-pip since that 'experiment' has been discarded (after being pushed into the 'stable' branch on 10/10 - anyone who had this issue before then was on the unstable or testing branches, and should have expected problems anyway).

ivankravets, 2019-10-12:
@pfeerick thank you so much for the professional help with reproducing this issue. It's not a problem for us to avoid this issue. I'm this `pip.conf` will help to avoid similar issues later.

I also understand Manjaro maintainers. They want to avoid issue when users install Python packages into the system `site-packages` location.

The new release 1.9.1 will be released soon. We did a lot of critical changes in new installer. Need to wait for other bug reports and fix them especially on Windows. We switched to Python 3 for Windows users if they don't have own Python.

pfeerick, 2019-10-12:
No problem... gets another issue closed :slightly_smiling_face:  And yes, ignorant `sudo pip` has had some... disastrous consequences... but this just wasn't the way to fix it. 

As I mentioned earlier, I installed 1.9.0 successfully on Windows 10 (no system python at all) as well as Ubuntu (stock python3 IIRC) with no issues. I'll update my other Windows 10 box tomorrow probably, so if you don't hear from me that went well also ;)

ivankravets, 2019-10-12:
Thanks!

 P.S: We work on something super cool that is coming to PlatformIO  :) I'm sure you will love it.

ivankravets, 2019-10-19:
1.9.1 is out! Please update. 
- https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.9.1

closed 2019-10-19 by ivankravets