platformio/platformio-vscode-ide #1008: Start PIO Home Server

jimmy85R, 2019-09-08:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/jimmy/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/jimmy/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-09-16:
Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

acmdorio, 2019-09-23:
> Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

First of all: thanks for your nice and very useful work !!!

I've got the same problem @ jimmy85R has declared... and after followed your orientation, the problem remains.

I've got python 3.7.3 (with pyenv).

Just to mention, my installed VSCode works well with other extensions.

Again, thanks, man!

ivankravets, 2019-09-23:
@acmdorio Have you had PlatformIO installed before updating to macOS Catalina? Did you use Python 3 before on this machine?

acmdorio, 2019-09-23:
> @acmdorio Have you had PlatformIO installed before updating to macOS Catalina? Did you use Python 3 before on this machine?

Yes, my installation of VSCode was before the upgrade. On the other issue, I recently upgrade to v.3.7.3...  at that time (VSCode installation) I had the system version only (v.2.7).

Any hints?
Thanks!

ivankravets, 2019-10-10:
Do you use `brew`? Did you overwrite official Python with a customer before 10.15 update? If yes, please repeat this procedure (reinstall custom Python). Your custom installation is broken.

See the solution here https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-10 by ivankravets