platformio/platformio-vscode-ide #1010: Start PIO Home Server

robobino, 2019-09-08:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/robwaltz/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/robwaltz/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-09-16:
Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

alexpb77, 2019-09-18:
> Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

Yes, I'm removed all under ~/.platformio/penv, but still the same error persists. 

```
~  python --version
Python 3.7.4

```

Temporary downgrade to 1.8.0 - it works.

ivankravets, 2019-09-20:
Do you have `python2 --version`?

thalyssra, 2019-09-25:
Catalina has python 2.7.16 and python 3.7.3.

```
~ ❯❯❯ python --version
Python 2.7.16
~ ❯❯❯ python2 --version
Python 2.7.16
~ ❯❯❯ python3 --version
Python 3.7.3
```

I'm experiencing the same thing here, and came to read through the bugs to see if anyone had made any progress.  Let me know if there's anything I can do to help, though I'm not extremely familiar with the process and am just trying to learn PlatformIO

ivankravets, 2019-09-25:
> Catalina has python 2.7.16 and python 3.7.3.

No, see https://apple.stackexchange.com/questions/361840/when-updating-to-macos-catalina-will-the-default-python-installation-get-removed

It seems like you installed them before on macOS 10.14. Did you install Python 2.7.16 with `brew`? Please provide an output from `brew list`.

thalyssra, 2019-09-25:
```
~ ❯❯❯ brew list
adns		bison		flex		gnu-sed		htop		libgpg-error	libssh2		lua		mr		openssl@1.1	pkg-config	tmux		zlib
autoconf	cmake		gawk		gnupg		jpeg		libidn		libtasn1	lzip		ncurses		p11-kit		readline	tokyo-cabinet
automake	coreutils	gdbm		gnutls		libassuan	libidn2		libtool		lzo		nettle		pcre		socat		unbound
bash		crosstool-ng	gettext		go		libevent	libksba		libunistring	m4		nmap		pcre2		sqlite		vde
bettercap	cscope		git		gpgme		libffi		libpng		libusb		make		npth		pinentry	swig		wget
binutils	emacs		gmp		grep		libgcrypt	libssh		lmdb		mpfr		openssl		pixman		talloc		xz
```

Regaron, 2019-10-09:
Installing python2 from python.org helped solve this.

ivankravets, 2019-10-10:
Do you use `brew`? Did you overwrite official Python with a customer before 10.15 update? If yes, please repeat this procedure (reinstall custom Python). Your custom installation is broken.

See the solution here https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-10 by ivankravets