platformio/platformio-vscode-ide #1012: Start PIO Home Server

ofilin, 2019-09-11:
%23 Configuration

VSCode: 1.38.0
PIO IDE: v1.8.3
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: Error: Could not find a version that satisfies the requirement '~2.34.190418' for your system 'darwin_x86_64'

	at S.Promise (/Users/user/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/user/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

Schaggo, 2019-09-14:
same here on macOS 10.15

ivankravets, 2019-09-16:
Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

karim1221, 2019-09-27:
where is the folder ~/.platformio/penv?

Schaggo, 2019-09-27:
I updated python and everything worked.

anh-tuan, 2019-09-27:
> I updated python and everything worked.

How can you update? Please give me a hint. Thanks

ivankravets, 2019-09-27:
@anh-tuan 
Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

anh-tuan, 2019-09-27:
OK this is my error:

Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

Exception
```
Error: 
	at S.Promise (/Users/tuann/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/tuann/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```
$PATH: /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Library/Apple/bin:/Applications/platform-tools:/Applications/VMware Fusion.app/Contents/Public

python --version
Python 2.7.16

pip --version
zsh: command not found: pip

ivankravets, 2019-09-27:
It seems like you manually installed Python 2.7.16. How you did that? Via `brew`? Just uninstall this Python 2.7.16 and install again.

anh-tuan, 2019-09-27:
I remember that I didn't install custom version of python, maybe it has come with other software.

Let me try to uninstall it. Which version is default? @ivankravets

Schaggo, 2019-09-27:
MacOS comes with 2.7.16. 
After installing 3.x you have to disable 2.7 

This nice gentleman mad a tutorial

https://wsvincent.com/install-python3-mac/

ivankravets, 2019-09-27:
Just download 2.7.16 installer https://www.python.org/downloads/release/python-2716/ and reinstall.

Remove `~/.platformio/penv` folder and start PIO Home.

ivankravets, 2019-09-27:
> MacOS comes with 2.7.16.

With 2.7.10.

ivankravets, 2019-09-27:
See https://opensource.apple.com/source/python/

ivankravets, 2019-09-27:
This issue is linked only with machines which have multiple Pythons in a system: 2.7.10. 2.7.16. etc. It seems like custom Python was not properly migrated with upgrading to 10.15

anh-tuan, 2019-09-27:
But I found nothing to uninstall in /Library/Frameworks :(

anh-tuan, 2019-09-27:
2.7.16 is default version of 10.15:

WARNING: Python 2.7 is not recommended.
This version is included in macOS for compatibility with legacy software.
Future versions of macOS will not include Python 2.7.
Instead, it is recommended that you transition to using 'python3' from within Terminal.

Python 2.7.16 (default, Jul 14 2019, 03:47:49)
[GCC 4.2.1 Compatible Apple LLVM 11.0.0 (clang-1100.0.32.4) (-macos10.15-objc-s on darwin
Type "help", "copyright", "credits" or "license" for more information.

ivankravets, 2019-09-27:
Stop, could you contact me@ikravets.com? I'll need more info. Please don't install or uninstall any software. I need to reproduce this issue. To be honest, I don't understand the root of the problem. Thanks!

closed 2019-09-27 by ofilin

anh-tuan, 2019-09-28:
[image]

Following your guide, I uninstall Python 3, `rm -rf ~/.platformio/penv`, it works like magic!

Honestly, you're right @ivankravets , the root cause can come from multiple Python versions.

Thank you so much.

ivankravets, 2019-09-28:
What is your Python version now?
```
python --version
```

anh-tuan, 2019-09-28:
```
python --version
Python 2.7.16
```

Schaggo, 2019-09-28:
I just checked I have both