platformio/platformio-vscode-ide #1013: Google drive file stream and Platformio don't like each other

craigdetter, 2019-09-11:
When trying to open a PlatformIO project hosted in a Google Drive File Stream folder, most of the time, PlatformIO isn't enumerating the files on the drive.  The entire source folder is set to "Available Offline." The platformio.ini is in the root of the project folder (see attached image).

[image]

Versions of software:
MacOS Mojave 10.14.5
Google Drive File Stream: Version: 33.0.16.0
PlatformIO: Home 2.3.3·Core 4.0.3

ivankravets, 2019-10-11:
Is it PlatformIO issue? We just work with your file system.

craigdetter, 2019-10-11:
Yes, It's a PlatformIO issue. It should work with Google Drive (which is a
very popular file system).

ivankravets, 2019-10-11:
How to fix it? What PlatformIO does wrong? PlatformIO is written in Python and uses its pathlib.

craigdetter, 2019-10-11:
Create Google Drive (setup a Google Email (free), comes with small Google
Drive file space like 15Gb (for free),
Test putting the project source files on Google Drive:
- put the Platform.io project on the google drive,
- then open the project, close it, then reopen it a few times

pfeerick, 2019-10-15:
I don't see how this is a Google Drive issue... if you have it set to 'Available offline' then the project folder is nothing more than a folder on your local device which is keep in sync with the cloud service. I could see there being problems if it were set to 'online only' as that would require the Google Drive to make the files available as accessed. 

It's more likely the issue with VSCode (as that is what is responsible for showing the File Explorer panel) needs the panel refreshed, or the folder is collapsed. Move over the file explorer panel, and then hit the reload icon up the top to see it that fixes it. 

[image]

craigdetter, 2019-10-16:
I'm not even getting to VSCode, I'm at the PlatformIO Open Project Window,
see attached.

pfeerick, 2019-10-17:
What about `File -> Open Folder` (since that is basically the same thing). And it looks like email attachments don't work, as nothing showed up.

pfeerick, 2019-10-18:
That suggests it is a Python/PlatformIO issue... as if VSCode can see it... PlatformIO should be able to also. Does the Google Drive client do anything fancy with file system links on Mac? Does anything change when you click the refresh icon in the `Open PlatformIO Project` file browser? (maybe it doesn't enumerate the folder contents properly the first time?) Try the 'Show Hidden Files` third icon along) as well to see if that changes anything...

[image]

craigdetter, 2019-11-08:
Responses to your questions:
I think the primary issue is that GoogleDrive doesn't show up under "Devices".

That is, using VS Code -> file open, the Finder opens where "Google Drive" appears under "Locations" which is the equivalant to "Devices" in PlatformIO. However, Google Drive doesn't appear in "Devices."

pfeerick, 2019-11-09:
Can't really comment on that from experience, but unless Mac Mojave + Google Drive is doing something really weird... I wouldn't have thought that made any difference, since it's just really just a folder (although it's being synced with a online service). 

The more important question is "do the files/folders show up if you press the reload button?". Also, try updating PIO since 4.1 has been released, and PIO Home has also been updated.

closed 2020-09-18 by stale[bot]