platformio/platformio-vscode-ide #1016: Start PIO Home Server

Schaggo, 2019-09-14:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/marcoschikatis/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/marcoschikatis/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

Schaggo, 2019-09-14:
updated python via homebrew. problem solved

janeoa, 2019-10-10:
> updated python via homebrew. problem solved

U mean like install the python3? or what?

ivankravets, 2019-10-10:
@janeoa Do you use `brew`? Did you overwrite official Python with a customer before 10.15 update? If yes, please repeat this procedure. Your custom installation is broken.

janeoa, 2019-10-10:
> @janeoa Do you use `brew`? Did you overwrite official Python with a customer before 10.15 update? If yes, please repeat this procedure. Your custom installation is broken.

I use brew, but did not use python after clean Catalina Beta install. Because I don't usually write in python, I don't really know what to update. So should I reinstall python (2 or 3) via brew?

ivankravets, 2019-10-10:
Do you have `python3 --version` in a system? Which version?

janeoa, 2019-10-10:
> Do you have `python3 --version` in a system? Which version?

Python 3.7.3

ivankravets, 2019-10-10:
@janeoa Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

P.S: Привет :) Да, все свои! :)