platformio/platformio-vscode-ide #1018: Start PIO Home Server

alexpb77, 2019-09-16:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/alexpb/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/alexpb/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-09-16:
Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now?

thalyssra, 2019-09-19:
I'm pretty sure this is due to Catalina's libcrypto.dylib crashing python because it doesn't allow linking to the non-versioned dylib anymore.

Previously, you would get a warning, that it shouldn't be done, but it appears that they've now set it to be an error if you do it.

`Invalid dylib load. Clients should not load the unversioned libcrypto dylib as it does not have a stable ABI.`. https://forums.developer.apple.com/thread/119429 - something along these lines

janeoa, 2019-10-10:
> ~/.platformio/penv

This worked! after updating python and using this command it booted

ivankravets, 2019-10-10:
We are working on a better solution for this issue and new version of IDE is coming soon.

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

closed 2019-10-11 by ivankravets