platformio/platformio-vscode-ide #1021: Start PIO Home Server

cheezebot098, 2019-09-18:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/michaelpersson/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/michaelpersson/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-09-20:
Could you remove `~/.platformio/penv` folder and restart VSCode. Does it work now? What is your `python --version`?

ivankravets, 2019-10-10:
Do you use `brew`? Did you overwrite official Python with a customer before 10.15 update? If yes, please repeat this procedure (reinstall custom Python). Your custom installation is broken.

See the solution here https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-10 by ivankravets