platformio/platformio-vscode-ide #1023: platformio-ide.* not found

aster94, 2019-09-20:
I am having a strange problem after having just installed vscode+platformio

the platformio commands seems to be hard to find, I made a video to show the issue: https://streamable.com/cq1i3

if i run pio run from the terminal works:
```
pio run
Processing d1 (platform: espressif8266; board: d1_mini_lite; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_lite.html
PLATFORM: Espressif 8266 2.2.3 > WeMos D1 mini Lite
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
No dependencies
Retrieving maximum program size .pio\build\d1\firmware.elf
Checking size .pio\build\d1\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  32.4% (used 26572 bytes from 81920 bytes)
PROGRAM: [===       ]  26.9% (used 257696 bytes from 958448 bytes)
================================================================================================= [SUCCESS] Took 2.28 seconds ================================================================================================= 
```

but if i do F1 and choose platformio built i get 

> command 'platformio-ide.build' not found

output of $env:path:
`C:\Users\vincenzo\.platformio\penv\Scripts;C:\Users\vincenzo\.platformio\penv;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Users\vincenzo\AppData\Local\Microsoft\WindowsApps;;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Users\vincenzo\AppData\Local\Microsoft\WindowsApps; `

ivankravets, 2019-09-20:
Do you see PlatformIO icon in the left sidebar?

aster94, 2019-09-20:
if you mean the big one yes (I also put it as first one ❤)

[image]

ivankravets, 2019-09-23:
PlatformIO.ini should be in the root of project. Please open in VSCode `Software` folder directly.

closed 2019-09-23 by ivankravets