platformio/platformio-vscode-ide #1024: Continued frustrations getting Arduino IDE to be recognized in VSC

nkuck, 2019-09-20:
Never been able to get past the following message: "Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change."
I realize that it is my error in not knowing how to get the extension set up properly and addressing the Arduino/command path correctly, but I have tried everything that seems correct. Running on a MacOS.

ivankravets, 2019-10-11:
PlatformIO does not use Arduino software. Please uninstall Arduino extension from VSCode and don't install later if VSCode will ask you.

closed 2019-10-11 by ivankravets

nkuck, 2019-10-11:
Thanks, Ivan. If I may ask a very basic question: How do I load a simple Arduino sketch and also Platformio.ide. It seems that the latter file is necessary but not sure if I just search for it, which I have done, and open it, or should it load when I launch VSC? New to the VSC environment.

ivankravets, 2019-10-11:
Please open PIO IDE > PIO Home > Import Arduino Project.

nkuck, 2019-10-11:
But I don’t get the Home icon, only rarely, which tells me that I have done something basically incorrectly with the install??

ivankravets, 2019-10-11:
Did you install PlatformIO IDE extension in VSCode?

nkuck, 2019-10-11:
Yes

ivankravets, 2019-10-11:
What is your macOS version?
```
python --version
python3 --version
```

nkuck, 2019-10-11:
10.14.6 Sierra

ivankravets, 2019-10-11:
Try to remove `~/.platformio` folder and restart VSCode.

nkuck, 2019-10-11:
Same so far.

ivankravets, 2019-10-11:
Please make a screenshot from your VSCode. I don't know what you do.

nkuck, 2019-10-11:
Did you receive the screenshot?

ivankravets, 2019-10-11:
Please attach it here.

nkuck, 2019-10-11:
Does it matter that PlatformIO seems to be installed Globally?

ivankravets, 2019-10-11:
I don't know you do. I don't see any images. Please contact our community https://community.platformio.org/