platformio/platformio-vscode-ide #1030: Installation Manager

matt-alencar, 2019-09-25:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.37.1
PIO IDE: v1.8.3
System: Linux, 4.19.69-1-MANJARO, x64

%23 Exception
```
Error: PIP Core: ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.

	at f.value.o.runCommand (/home/matt/.vscode-oss/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:32222)
	at ChildProcess.h (/home/matt/.vscode-oss/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

hirnwunde, 2019-10-01:
same error here
Using VSCodium from AUR ([vscodium-bin 1.38.1-1](https://aur.archlinux.org/packages/vscodium-bin/)) instead of VSCode

# Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Linux, 4.19.69-1-MANJARO, x64

# Exception
```
Error: PIP Core: ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.

	at f.value.o.runCommand (/home/hwd/.vscode-oss/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:32222)
	at ChildProcess.h (/home/hwd/.vscode-oss/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1007

closed 2019-10-11 by ivankravets