platformio/platformio-vscode-ide #1031: Cannot set focus on library search text box.

brzezinol, 2019-09-25:
Issue Type: <b>Bug</b>

Click PlatformIO Home in status bar.
Go to Libraries.
Click on search text box, and focus not enter, I cannot search libraries.

When I do some tricks with left panels I sometimes can enter field.

Extension version: 1.8.3
VS Code version: Code 1.38.1 (b37e54c98e1a74ba89e03073e5a3761284e3ffb0, 2019-09-11T13:35:15.005Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (12 x 3696)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>oop_rasterization: unavailable_off<br>protected_video_decode: unavailable_off<br>rasterization: unavailable_off<br>skia_deferred_display_list: disabled_off<br>skia_renderer: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.91GB (9.08GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

VirgiliaBeatrice, 2019-09-26:
Same issue on my environments.
Under my investigation, the issue happens only when you click other tabs and then back to PIO Home tab.
As a temporary solution, you can close the PIO Home tab and reopen it again, then the focus of the text field will be back.
Another solution is just to minimize your VSCode window into taskbar once, and the focus will be back.

Anyway, hope that this bug can be solved soon.

Version: 1.38.1 (user setup)
Commit: b37e54c98e1a74ba89e03073e5a3761284e3ffb0
Date: 2019-09-11T13:35:15.005Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.18362
Extension Version: 1.8.3

thanxx, 2019-09-28:
+1 on the same issue

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/606

See solution https://github.com/platformio/platformio-vscode-ide/issues/606

closed 2019-10-11 by ivankravets