platformio/platformio-vscode-ide #1042: Installation Manager

diohabara, 2019-10-02:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Linux, 4.15.0-62-generic, x64

%23 Exception
```
Error: PIP Core: /home/denjo/.platformio/penv/bin/python: No module named pip

    at f.value.o.runCommand (/home/denjo/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/installer/stages/platformio-core.js:323:16)
    at ChildProcess.h (/home/denjo/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/misc.js:172:5)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
    at Socket.emit (events.js:182:13)
    at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-10-11:
What is your Python version? Do you have `python3 --version`?

diohabara, 2019-10-11:
Sorry to send this message without adding any additional information.

Yes, I have it. Here's what my terminal says.

```shell
$ python3 --version
Python 3.6.8
```

diohabara, 2019-10-11:
Plus, this is my OS environment.

```shell
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

closed 2019-10-18 by ivankravets